raspberrypi/documentation #148: Update hardware USB issues

gripet, 2015-01-13:
USB ports on all RPis (including Model B+) is unable to source sufficient current for powerful USB devices, such as WiFi dongles. Using external hub lowering advantages of 4-ports Model B+. The simple solution is delivering +5V to powerful devices bypassing Pi.

```
                   +-----------------------+
-----------+       |      +-----------+    |
           |-(+5V)-+   x--|           |    |
 500 mA    |-( D-)--------|    RPi    |----*--(+5V)
USB device |-( D+)--------|           |-------(GND)
           |-(GND)--------|           |
-----------+              +-----------+
```

JamesH65, 2015-01-13:
Whilst what you are suggesting might work, you should be able to source enough power from the Pi itself for the majority of USB devices. I've never had problem with WiFi on the B+, and can run a harddrive from it at the same time (You do need to enable high powered USB output by putting max_usb_current=1 in config.txt)

gripet, 2015-01-14:
Thank you for the info! But when using more than two hi-load dongles (this is my case) 1.2A is still unsufficient.

JamesH65, 2015-01-14:
I suspect that the overall power consumption of the device with such an
unusually high load would cause the input polyfuse to fuse.

I'm not convinced that the documentation need to be updated in this way
though. This sort of modification could cause unexpected problems for the
less tech savvy.

waveform80, 2015-01-14:
@gripet what, specifically, are the two (or more) high load dongles that you're using?

gripet, 2015-01-15:
It is not unusually high load, just standard 500 mA in peak per device. I'm using Netgear WNDA4100 802.11n adapters in monitor mode. I can plug several adapters and it seems working properly, but only if I'm actively using not all of it. When switching all adapters to monitor mode, the Pi's current limiter drops the voltage, and it can be seen some strange effects, like: some of the interfaces going down, device reconnects, disappear at all, etc. 
Pi's polyfuse is bypassed in my case. There is no problem with it. I posted this issue just for information, may be not need to update documentation, but my solution may be useful for somebody.

JamesH65, 2015-01-15:
Have you tried a better USB lead? And what power rating is the supply?

gripet, 2015-01-15:
Yes. Power supply is 5V 3A, and it's of sufficient quality (I tested it).
Do you think something is wrong? This behavior is expected, when adapters want to consume more power, than Pi's current limiter can supply, there was some troubles. When using proper powering now my system is stable. 
It's not good idea to limit the current, I think it's better to latch it off (how good hubs do). This method may help to localize the problem, instead of masking it.

JamesH65, 2015-01-15:
I certainly wouldn't expect just a couple of (high power) devices
overloading the device. I use a Wifi adapter as a AP and have a connected
HD, running from a car 12v->USB adapter and it works fine.

The Pi itself takes about 750mA when running full tilt, with 500mA per
devices that another 1000mA, total 1750mA, which should be fine, or are you
using more adapters than that? I'm not sure what the total the Pi can
supply is though, I'll find out.

P33M, 2015-01-15:
The USB switch is a current-limiting device as this is necessary to prevent dangerous overloads. The current limit is suppressed for 7mS at the commencement of any overload. The action taken by the hub on signalling of current limit by the USB switch is to latch off the power after 20mS of current limit. The 20mS is to act as a de-bounce interval in order to be immune connection of devices that have inrush current.

A message is printed to the kernel message log saying that an overcurrent has been detected. Lack of this message == no overcurrent event.

Ganged power switching is a valid method for switching power to USB ports. It is defined in the USB specification. The 1.2A ganged limit is both intentional and a valid choice. If you want to connect a device (or multiple devices) then a powered hub is required.

The hardware undervoltage detection (~4.63V) is tied the PWR LED on the Pi, if this light ever extinguishes or blinks in operation when USB devices are attached to the Pi then all bets are off, as the voltage to downstream devices has gone below the specified minimum.

gripet, 2015-01-15:
> I certainly wouldn't expect just a couple of (high power) devices overloading the device

Why not? Couple of dongles may cause overload when limit is 0.6A (1A required), three ones if limit is 1.2A (1.5A required). It depends on device type. May be low average consumption and high only in peak (and peaks from several devices may not correlate), may be constantly high.

gripet, 2015-01-15:
Thanks to all!

closed 2015-01-15 by gripet