raspberrypi/documentation #150: New config.txt syntax options in  ../configuration/device-tree.md

DougieLawson, 2015-01-21:
Now that the 3.18.3+ kernel with the device tree has gone into the wild there's a new syntax for the device_tree parameters in /boot/config.txt

This is what we've got today.
```# Uncomment some or all of these to enable the optional hardware interfaces
#device_tree_param=i2c1=on
#device_tree_param=i2s=on
#device_tree_param=spi=on

# Uncomment one of these lines to enable an audio interface
#device_tree_overlay=hifiberry-dac
#device_tree_overlay=hifiberry-dacplus
#device_tree_overlay=hifiberry-digi
#device_tree_overlay=iqaudio-dac
#device_tree_overlay=iqaudio-dacplus

# Uncomment this to enable the lirc-rpi module
#device_tree_overlay=lirc-rpi

# Uncomment this to override the defaults for the lirc-rpi module
#device_tree_param=gpio_out_pin=16
#device_tree_param=gpio_in_pin=17
#device_tree_param=gpio_in_pull=down
```

The docs need to have 
```
device_tree=
``` 
which disables the device tree

It needs
```
dtparam=i2c1=on,i2s=on,spi=on
dtoverlay=lirc-rpi
dtoverlay=lirc-rpi,gpio_out_pin=17,gpio_in_pin=13
```
added as valid syntax options. Which are easier to use and more compact and should probably be listed as the preferred way of doing this stuff.

mati75, 2015-01-28:
Add to /boot/config.txt line:
```
device_tree_overlay=overlays/w1-gpio-pullup-overlay.dtb
```
solved this issue.

pelwell, 2015-02-23:
Hey, wadda ya know, I've got round to this now. See what you think.

lurch, 2015-02-23:
Looking really good :-)
Unfortunately the side-effect of making the documentation so comprehensive :+1: , is that's it's become a bit *huge*. I wonder if it'd be possible to split it into 'user documentation' (editing config.txt etc.) and 'developer documentation' (writing overlays, using dtc etc.) ? Only a suggestion.

pelwell, 2015-02-23:
Hmmm, yes, perhaps. But let me have some fun writing software first.

lurch, 2015-03-02:
@DougieLawson Can this be closed now? Or is there additional info that http://www.raspberrypi.org/documentation/configuration/device-tree.md still needs?

DougieLawson, 2015-03-02:
You've still not documented all of the possible options and their syntax (you can code it in BNF if you like or with "railroad" diagrams) for a generic case (so that someone with a new device and a new device tree knows what's possible). 

There's also the variety of syntax options to do the same thing like: device_tree_param vs dtparam (if the former is still valid in the code). 

In my ideal world every parameter that can ever be coded in /boot/config.txt would have a complete syntax diagram and complete documentation for every keyword and option (perhaps, I've had too many years working with IBM mainframes, where things get documented like: http://www-01.ibm.com/support/knowledgecenter/SSEPH2_13.1.0/com.ibm.ims13.doc.sdg/ims_dfsvsmxx_proclib_service.htm scroll down that page and enjoy a beautifully crafted "railroad" diagram).

I think you also need to pull Phil's /boot/overlays/README into the main docs. It's too well hidden for the novice users where it is now.

pelwell, 2015-03-03:
1) No useful features are undocumented.
2) No features have been removed, but the longer old names are no longer documented, and therefore effectively deprecated, to reduce confusion.
3) In the near future the README and the sources to the overlays will be colocated, in order to encourage their simultaneous updating. At that time, the README may be split into multiple files. The existing documentation does refer to (and link) to the README, but it could be more prominent.

lurch, 2015-03-03:
@DougieLawson 
> In my ideal world every parameter that can ever be coded in /boot/config.txt would have a complete syntax diagram and complete documentation for every keyword and option

Unfortunately AFAIK the RPF have yet hired a team of full-time technical authors to work on the documentation ;-)

JamesH65, 2015-03-03:
Seems to me that would be a good job for someone outside the Foundation to
do and then submit a PR. Means the RPF can get on with the complicated
stuff with the limited resources they have.

DougieLawson, 2015-03-03:
OK. I'm happy with the current stuff.

If you wanted to give me access to the source code for the firmware blob (with an NDA) and pay a reasonable hourly rate I'd write up all the parms in excruciating detail.

closed 2015-03-03 by DougieLawson