raspberrypi/documentation #155: References to "rpi-update" as a debugging step

Gadgetoid, 2015-01-29:
I think rpi-update has permeated the Pi community far too deeply, and is treated as the "recommended" way to fix bugs, and listed as a debugging step in far too many places.

The fact is, rpi-update is just as likely to cause problems as it is to fix them, and it's very much a hamfisted and bruteforce attempt to problem solving.

Admittedly, and much to my chagrin, I have found myself suckered into recommending it as a solution to bugs. However, most of those bugs were, in fact, caused by running it in the first place ( the mysterious disappearance of i2c last year for example ).

I think references to rpi-update in articles such as this: https://github.com/raspberrypi/documentation/blob/master/troubleshooting/hardware/camera.md

should be removed altogether, and it should be de-emphasized as a "not recommended" way of grabbing bleeding-edge kernel builds in more advanced documentation. 

Rpi-update's days are numbered, and It's becoming clear that an apt repository is probably a better, if not the right, way to distribute such firmware, although I don't believe there's one available in-sync with rpi-update yet: http://www.raspberrypi.org/forums/viewtopic.php?f=66&t=80023

Basically, this tool looks, sounds and acts too much like an official way to upgrade to stable firmwares and has found its way into far too many guides. Most Pi users shouldn't even need to know it exists, much less be running it on a regular basis under the assumption that the "bleeding edge" is of any use to them whatsoever.

JamesH65, 2015-01-29:
I'd disagree it creates as many problems as its solves. Many people run it
every day and see now problems.

But I do agree it should be de-emphasised.

lurch, 2015-01-29:
Just to try and keep everything tied together, in addition to the multiple comments on #153, I've also just spotted https://github.com/Hexxeh/rpi-update/issues/158

But this issue is probably the best place for the 'meta-discussion' moving forwards?

ebenupton, 2015-03-04:
Closed in previous PR.

closed 2015-03-04 by ebenupton