raspberrypi/documentation #156: Python FAQs

elParaguayo, 2015-01-29:
Hi,

I'd like to add an FAQ page for python as we see the same questions over and over again in the forum.

More here: http://www.raspberrypi.org/forums/viewtopic.php?f=32&t=97816

If the idea is ok then I'm happy to have a go at this.

bennuttall, 2015-01-29:
I like the idea of technical FAQs - we've been discussing something similar recently. Breaking up into sections would make sense too.

Let me check with the team and see what everyone thinks.

If I haven't responded by mid- next week feel free to bump this issue to remind me.

Thanks

elParaguayo, 2015-02-08:
Ben,

Bumping as requested!

JamesH65, 2017-03-06:
@ben More bumping - has this been done?

JamesH65, 2017-04-25:
@bennuttall More bumping - any progress on this?

bennuttall, 2017-04-25:
No - I don't think this belongs in documentation.

JamesH65, 2017-04-25:
OK, closing.

closed 2017-04-25 by JamesH65