raspberrypi/documentation #158: Document the bcm_host helper functions

P33M, 2015-02-05:
For programs wanting to poke peripheral addresses directly (for lack of a better interface), they need to know where in the virtual memory map the peripheral bus segment has been placed. This used to be tied to the platform, now is variable between Pi1 and Pi2.

Perhaps we need a one-pager in the hardware section detailing how to use this api:

https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/bcm_host.h

ghollingworth, 2015-02-05:
:)

ebenupton, 2015-03-04:
Jonathan - can you propose some text, and where you think it should be located in the documentation?

JamesH65, 2017-03-02:
Hi @p33m, any news on this?

JamesH65, 2017-04-20:
@P33M @popcornmix If either of you guys can give me a leg up on how this stuff works and why it's needed, I'm happy to write the documentation for it. Would be nice to get this 2 year old issue out of the way!

popcornmix, 2017-04-20:
The code is here:
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c#L150-L166
They just read from device-tree (falling back to hard-coded Pi1 addresses if not available) the addresses needed for accessing sdram or peripherals from arm.

bcm_host_get_peripheral_address
Gets the arm side physical address of where peripherals are mapped.
Will be 0x20000000 on Pi1 and 0x3f000000 on Pi2

bcm_host_get_peripheral_size
Gets the size of peripherals space
Will always be 0x01000000

bcm_host_get_sdram_address
Gets the bus address of usual alias of sdram.
Will be 0x40000000 on Pi1 (GPU L2 cached) and 0xC0000000 on Pi2 (uncached).

These are of interest to very few users - i.e. those who want to mmap peripherals to access without using a kernel driver.

JamesH65, 2017-04-21:
@popcornmix Does this read OK for the bcm stuff? Which library do you need to build in to get this library, just bcmhost?

JamesH65, 2017-04-21:
Sorry, I should have linked to the PR. #623

PR #623 (2017-05-08): James h65 patch 1#158
Add a new peripherals address page

Fixes #158
closed 2017-05-08 by JamesH65