raspberrypi/documentation #161: Device tree documentation

framboosejh, 2015-02-12:
The device tree documentation says a lot about activating or diactivating the device tree and even how to enable I2C, SPI and so on, but it says nothing over the fact that some entries in /sys are changed. by example that '/sys/devices/virtual/gpio/gpio' is moved to  "/sys/class/gpio/gpio" or even to a different place. I spend 4 days with looking why my Piface Digital doesn't work anymore. First i thought it was a problem with the device tree and had to put on SPI on with dtparam=spi=on, but that didn't work. In the end i found that you also have to change the directory to /sys/class/gpio/gpio25.

This all is very bad documentated. And there is no place on internet where the device tree for the raspberry is explained. I have the impression that the device tree is implemented had over heels in the kernel and that nobody cares about all the frustated users who's add on boards don't work anymore after an upgrade.

Yes big news the release of the raspberry 2, and i did buy one to. It was an reason for me to abandon my Beagle bone black in favor of the rpi 2. But when the support of the device tree remains so poor i go back to the BBB, at least they have a much better implementation of the device tree and documentation.

fivdi, 2015-02-12:
@framboosejh There's more information about this issue [here](https://github.com/raspberrypi/linux/issues/791).

pelwell, 2015-02-12:
/sys/class/virtua/gpiol/gpio is the wrong location. /sys/class/gpio was there before, and will be there in the future, because it is the right place.

Some things about Linux aren't documented very well, so people make up there own ways to do them. Other things are documented very well (https://github.com/raspberrypi/linux/blob/rpi-3.12.y/Documentation/gpio.txt, https://github.com/raspberrypi/linux/blob/rpi-3.18.y/Documentation/gpio/sysfs.txt), but people come up with their own ways of doing things anyway. Switching to using Device Tree has changed some of the implementation details, but the major interfaces - the ones people should have been using - are still the same.

Stuff gets broken with every kernel release - that's why at Raspberry Pi we encourage advanced users before each major update to try "BRANCH=next rpi-update" and make sure it works for them. The "New Device Tree Kernel in testing" thread was started 2 1/2 months ago. You might want to get a preview of the new developments next time, and help us shake out the wrinkles before it hits the mainstream.

lurch, 2015-03-02:
There's now more comprehensive DeviceTree documentation at http://www.raspberrypi.org/documentation/configuration/device-tree.md

@framboosejh Can this issue be closed now?

closed 2015-03-04 by ebenupton

framboosejh, 2015-03-07:
Yes,
Okay that the issue is closed, I have to study the new documentation first and that will take time, before i can ask new questions.
Thank you for the effort and time you put in it. It’s a shame i didn’t react earlier on github, i could have said this there, so that other people knew that you helped me very well.

Evert Jan Henken