raspberrypi/documentation #164: GPIO documentation needs updating for B+ and later models

DeeJay, 2015-02-18:
https://github.com/raspberrypi/documentation/blob/master/usage/gpio/README.md needs updating

"Seventeen of the 26 pins are GPIO pins; the others are power or ground pins."

lurch, 2015-03-02:
Given the prevalence of Models A+ / B+ and Pi 2, I guess all the images ought to be updated to show the 40-pin header too? Perhaps using some of the templates Sam made for the education resources?

ebenupton, 2015-03-04:
Perhaps ask @clivebeale to do some rewrites and then move this somewhere more appropriate.

clivebeale, 2015-03-07:
Done. Now two guides -- started to write as one but too many differences in text and graphic and became messy and over-complex. @bennuttall : GPIO now has two guides, one for A and B and one for Plus models and Raspi2.

lurch, 2015-03-11:
`usage/README.md` currently links to the 'original' `gpio/README.md` but doesn't yet link to the new `gpio-plus-and-raspi2/README.md` (and AFAIK nothing else does either).
I won't submit a PR, because I dunno how you want to handle to linking to two different versions of the same guide (which I guess has a small overlap with #182).

Re-opening so that this doesn't get forgotten about.

clivebeale, 2015-03-11:
I've just split it into 2 links, unless anyone else has any better ideas :)

closed 2015-03-11 by clivebeale

bennuttall, 2015-03-11:
That's fine - perhaps add a link at the end of the first one linking to the second?