raspberrypi/documentation #169: Update instructions for new GUI

bennuttall, 2015-02-18:
- [ ] [Scratch](https://github.com/raspberrypi/documentation/tree/master/usage/scratch/README.md)
- [ ] [Python](https://github.com/raspberrypi/documentation/tree/master/usage/python/README.md)
- [ ] [Sonic Pi](https://github.com/raspberrypi/documentation/tree/master/usage/sonic-pi/README.md)
- [ ] [Minecraft](https://github.com/raspberrypi/documentation/tree/master/usage/minecraft/README.md)
- [ ] [Python Games](https://github.com/raspberrypi/documentation/tree/master/usage/python-games/README.md)
- [ ] [Mathematica](https://github.com/raspberrypi/documentation/tree/master/usage/mathematica/README.md)
- [ ]  [WiFi Configuration](https://github.com/raspberrypi/documentation/tree/master/configuration/wireless/README.md)

bennuttall, 2015-02-18:
Please add comments to alert us of more

lurch, 2015-02-18:
Whilst I agree that the documentation ought to be updated to cover the 'new' GUI, I wonder what this means for all the people still using the 'old' GUI?
Does the documentation need to cover both desktop interfaces; or should it only document the latest version and assume that everybody has already upgraded / will upgrade ?

bennuttall, 2015-02-18:
I think "Open from the Desktop or applications menu" makes sense. Users of the old GUI will have icons on the desktop, users of the new GUI will have it in the menu and have the ability to add icons to the desktop (we're advising teachers do this for their regular applications).

I'm just adding this text and screenshots of the menu.

lurch, 2015-02-18:
Great, best of both worlds :)
Do you need to specify *which* 'applications menu', or with the much-shorter menus that Simon introduced should we assume it'll be easy enough for the user to look through all the menus until they find the application they want?

pelwell, 2015-02-18:
Browsing is good - they might find something they haven't seen before.

bennuttall, 2015-02-18:
The screenshots should help

closed 2015-02-18 by bennuttall

lurch, 2015-02-18:
Looks like a couple of image tags are wrong / missing: https://travis-ci.org/raspberrypi/documentation

lurch, 2015-02-20:
I guess some of the resources need updating in a similar manner too, e.g. http://www.raspberrypi.org/learning/storytime/worksheet.md still refers to **IDLE3**.

Would it be possible to have a "central issue" (like this one was), to which all problems concerning the 'new gui' could be reported; or would each issue need to be reported individually to the specific repo in question? (i.e. https://github.com/raspberrypilearning/storytime for the specific problem above)

commit 3730f11f (2015-02-18): Update instructions for new GUI, close #169