raspberrypi/documentation #171: Warning people about HDMI-to-VGA adapters

PeeterAngelo, 2015-02-19:
In **setup/monitor-connection.md** there is a mention of using an HDMI-to-VGA adapter for VGA display devices. However, not all adapters are safe to use (the unpowered ones, to be precise). A lot of adapters that are available on the market draw a huge amount of current from the HDMI port to power themselves - some even up to 1A, which is much more than the HDMI standard allows (50mA). While most HDMI-capable devices don't seem to mind, plugging in said adapter that draws more than 200mA will burn out a certain diode (near the HDMI port, marked D1), because the Pi isn't (due to manufacturing costs I believe) built to handle this.

These adapters *may* cause even more damage.

I'd recommend appending the section with a suggestion to use a **powered** HDMI to VGA adapter along with a warning that using an unpowered adapter may damage the Raspberry Pi.

There are threads on the forums documenting that several people have ruined their Pis with these unpowered adapters.

ebenupton, 2015-03-04:
Please propose a pull request.

lurch, 2015-03-04:
Fixed with #189

closed 2015-03-04 by lurch