raspberrypi/documentation #175: Device tree documentation/configuration/device-tree.md - missing some docu on the integer types usable for device-tree-parameters

msperl, 2015-02-22:
I have read somewhere that one can define different types of device-tree-parameters into structures:
int_16 separated via ";" and int_32 separated via ":", int8 ...

Unfortunately I can no longer remember that link (on the forum) that contained this, so please document this "officially" so that one can make use of it and it is easy to reference.

Thanks, Martin

pelwell, 2015-02-22:
It is on the list of things to document. In the meantime, here is the relevant post:

http://www.raspberrypi.org/forums/viewtopic.php?p=696581#p696581

lurch, 2015-02-22:
It's also possible to find a link to that forum post by looking through the recent firmware [commit-messages](https://github.com/raspberrypi/firmware/commits/master) and spotting
> firmware: arm_loader: Add support for 8-, 16- and 64-bit parameters
See: http://www.raspberrypi.org/forums/viewtopic.php?f=29&t=93015&start=325#p696581

on https://github.com/raspberrypi/firmware/commit/593aa0fbbca3d1fc02138781c82daf5bf955080d

pelwell, 2015-02-23:
I've updated the documentation; I hope it now explains things sufficiently.

msperl, 2015-02-24:
Thanks - looks good!

closed 2015-02-24 by msperl