raspberrypi/documentation #181: Documentation for config.txt says DeviceTree is unsupported!

lurch, 2015-03-02:
Creating this as an issue, as requested by @bennuttall :)

http://www.raspberrypi.org/documentation/configuration/config-txt.md still says:
> `device_tree`
Specifies a device tree filename on the boot partition. This is not officially supported.

...which is obviously no longer true! ;-)

@popcornmix does the `device_tree_address` option still exist?

@pelwell As there are additional DeviceTree-related parameters (`dtparam`, `dtoverlay`, any others?), I guess the best option would be to simply refer people directly to http://www.raspberrypi.org/documentation/configuration/device-tree.md ?

pelwell, 2015-03-02:
Re: `device_tree` - Although we support DTs, and we currently allow DT support to be disabled, do we want to officially support specifying an alternate DT blob?

`device_tree_address` still exists, but I've added `device_tree_end` to allow for automatic padding of the DTB to create space for any overlays. Looking at the code now I can see that specifying only the start address wouldn't work, so I need to fix that, but do we want to support this kind of tweaking?

The currently recommended DT-related options are just the ones listed - `device_tree`, `dtoverlay` and `dtparam`. If you want to be complete then we could mention that `device_tree_overlay` and `device_tree_param` exists as aliases, but I'm happy for them to be forgotten about.

lurch, 2015-03-02:
> ..., but do we want to support this kind of tweaking?

I've got no idea, that's something for @pelwell @popcornmix and @ghollingworth to discuss and decide... ;-)

ebenupton, 2015-03-04:
Andrew - please just make the "device tree" link in that para refer to 

http://www.raspberrypi.org/documentation/configuration/device-tree.md

commit 3566028b (2015-03-10): Point to separate device-tree documentation for config.txt

Fixes #181
closed 2015-03-10 by lurch