raspberrypi/documentation #182: Add Pi2 details and specs

lurch, 2015-03-02:
Just like #132, but for the Raspberry Pi 2 Model B ;)

Also, where things used to say "Model A", "Model B+", etc. should these now be updated to "Pi1 Model A", "Pi1 Model B+", or is there some other official nomenclature that should be used?  Something to add to https://github.com/raspberrypilearning/creating-resources/blob/master/style.md ?

lurch, 2015-03-02:
Also needs adding to http://www.raspberrypi.org/documentation/hardware/raspberrypi/models/

ebenupton, 2015-03-04:
Ben to clean up

lurch, 2015-03-20:
After seeing #199 I went looking for the "model specs" page on http://www.raspberrypi.org/documentation/ and was confused when I couldn't find it. I eventually discovered that I couldn't find it because of https://github.com/raspberrypi/documentation/commit/f68bfc961423aa200bb7fea610f0a255bf32f73a ;-)

I can see how `hardware/raspberrypi/models/README.md` (originally added by @JamesH65 ) is a bit of a duplication of http://www.raspberrypi.org/products/ , but IMHO the `hardware/raspberrypi/models/specs.md` page **is** still useful, which now has no links to it. When Ben gets round to tackling this issue (I know he's still busy with resources stuff) maybe he could add a link to `hardware/raspberrypi/models/specs.md` from somewhere else?

ebenupton, 2015-03-20:
I think the conclusion was to retire the specs page completely, and to move any useful information onto the main website product pages.

flashspys, 2015-03-23:
Is the Peripheral specification for the BCM2836 (Pi2's cpu) already somewhere in the internet? We need it because some adresses were changed.

P33M, 2015-04-09:
See https://github.com/raspberrypi/documentation/commit/b0639f5a3ff6ba9b5006c62d643403d97408b1f7 for details on how the Cortex-A7 cluster on BCM2836 has been implemented.

Other than that, Pi2 is basically identical to a B+ in terms of mechanical layout.

bennuttall, 2016-01-28:
Are we dropping specs or updating them? They ought to be on the product pages really...

ebenupton, 2016-01-31:
We should retire this page.

commit 2a46bba6 (2016-02-01): Remove models/specs pages, close #182
closed 2016-02-01 by bennuttall