raspberrypi/documentation #184: pip-3.2

paddywwoof, 2015-03-02:
any chance of changing pip-3.3 to pip-3.2 in 
documentation/linux/software/python.md
to make whole thing a little less confusing for people new to python (ideally raspbian would have called it pip3 to avoid the inevitable confusion which will happen once the switch to jessie happens (bugs.debian.org/release-critical/))

lurch, 2015-03-03:
Fixed, thanks. This doesn't have anything to do with Jessie, but it looks like addf098 was wrong in introducing mentions of pip-3.3
\#BlameNuttall ;-)

closed 2015-03-03 by lurch

bennuttall, 2015-03-03:
Yeah the use of `pip-3.3` just a mistake.

But yeah what you said about `pip`, `pip2` and `pip3` is correct. Hopefully we'll move to those soon or when we upgrade to Jessie.

commit fc630bcf (2015-03-03): Change mentions of pip-3.3 to pip-3.2

...as that's actually the version of pip installed into Raspbian with `sudo apt-get install python3-pip`.
Fixes #184