raspberrypi/documentation #188: Mechanical drawings etc for A+, Pi 2 and other new toys

ebenupton, 2015-03-04:
Sticking this in here so we don't forget.

ebenupton, 2015-05-07:
Dummy comment.

hultqvist, 2015-12-21:
I assume [MECHANICAL DRAWINGS](https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/README.md) is the page that needs updating.

I've read in a random forum thread that that RPi 2 uses almost the same machanical schematics as the Rpi 1 B+, if that is the case it would really help if a note about that was added to the page.

JamesH65, 2017-03-06:
@Roger-Thornton Roger, si there anything we need/can add to the mechanical drawings page? Or is it as complete as we can do?

JamesH65, 2017-03-09:
See also #517

lurch, 2018-02-04:
#642 claims to have closed this, but #642 didn't add a mechanical drawing for the A+ ?

stefanwendelmann, 2018-02-09:
Yes a mechanical drawing fpr A+ is still missing :(

lurch, 2018-02-09:
Just like #382 , perhaps this should be re-opened?

JamesH65, 2018-02-09:
Yes, it is missing, had a quick chat the other day with the person who
deals with this - its will get done, but we are extremely busy at the
moment with more important stuff. I'll reopen the issue.

JamesH65, 2018-07-02:
@Roger-Thornton Did we ever get the missing A+ drawing done?

matthieu-lapeyre, 2019-01-23:
I raise my hand as well to get a Raspberry Pi 3 A+ mechanical drawing with the precise position of each I/O port. 
It should not require more than a half an hour to get it done. I can help if you miss time to do it.

JamesH65, 2019-01-23:
OK, just uploaded the 3A+ (also applicable to A+) drawing to the github repo, should be online on the main website within an hour.

JamesH65, 2019-02-06:
I'm going to close this as its been around a bit; if there are missing diagrams, please start a new issue for each missing item.

PR #642 (2017-05-15): Update schematics and mechanical diagrams, close #95, close #188
Closes #95 and #188
commit bd090f76 (2017-05-15): Merge pull request #642 from raspberrypi/schematics-update

Update schematics and mechanical diagrams, close #95, close #188
closed 2019-02-06 by JamesH65