raspberrypi/documentation #193: B+ schematic documentation could use lables.

mf0hka, 2015-03-11:
For example, what is chip U3? switching power supply? there is no label, and no part number. 
(my B+ hasn't arrived yet, so I really have no clue.)
hey U4, is that a voltage reffrence? (or behaving like one) well, I mean, I could look it up, as the part number is there at least. 

Reffrence this document:
http://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/Raspberry-Pi-B-Plus-V1.2-Schematics.pdf

ok, I see the tag, reduced schematic, where is the full schematic?

lurch, 2015-03-11:
> where is the full schematic?

It hasn't been publicly released yet.

JamesH65, 2017-03-02:
No change to schematic policy, closing for now.

closed 2017-03-02 by JamesH65

GeorgeSapkin, 2017-05-10:
@JamesH65 What is this policy that keeps being mentioned? It's nowhere to be found on schematics page. Would be nice to know why some of the schematics are missing (e.g. A+) and some of them are incomplete.

JamesH65, 2017-05-10:
Our policy is that the full schematics for are not going to be released for the foreseeable future.

GeorgeSapkin, 2017-05-10:
Can this be added to schematics page? Seems like this questions comes up a lot.

JamesH65, 2017-05-10:
I'll have a think whether the schematics pages need updating.