raspberrypi/documentation #195: USB Audio broken in kernel 3.18.7+

Halloran, 2015-03-15:
I run an RPi as an audio server: mpd 0.19.7 using ALSA to output to a Schiit Modi USB DAC, serving MP3's and FLAC files.

Recently, I ran apt-get update, apt-get upgrade, and my kernel went from 3.12.28+ to 3.18.7+, and this completely broke the audio output. Audio was being output more or less as white noise, not as slightly distorted, and this was the case with the aplay utility as well as with MPD. After a lot of troubleshooting with MPD, ALSA, etc., I used rpi-update to downgrade to the 3.12.28+ kernel, and all is working fine....

No idea it this is limited to USB audio, or if the problem might be deeper, but wanted to report it. Am happy to help with more system information, etc.

lurch, 2015-03-15:
Please report this at https://github.com/raspberrypi/linux/issues as this is a kernel issue rather than a documentation issue.

closed 2015-03-15 by lurch

Halloran, 2015-03-15:
Done. Sorry - my bad.