raspberrypi/documentation #196: Additions to quick start guide

turboscrew, 2015-03-15:
I think it would be a good idea to add pictures of Pi-versions with explanations in the quick start guide and/or documents. Something like this: 
[image]

I had to search the web to find out where (and which way) to insert the SD-card.
Even the video skips the attaching of cables and SD-card.
(BTW, the video kept stopping every few seconds - any idea why?)

Also the pictorial buyer's guide could be linked to help page.

bennuttall, 2015-03-15:
That's what the [Getting started with your Raspberry Pi](http://www.raspberrypi.org/help/what-is-a-raspberry-pi/) video was designed for, but yes I think we should add an updated labelled picture too.

lurch, 2015-03-15:
I guess this kinda overlaps with #41 too

bennuttall, 2015-03-15:
Yeah, I think once @samalderhyde is done with the resources we'll look at creating some general guides like this.

turboscrew, 2015-03-17:
In the video I really didn't seen the guy putting in the memory card (where, which side up).
About the video: it breaks a lot - about every 5 seconds. And it jams often. First at 00:18 then at 00:44 and so on. Clicking pause and then play made it run again, Sometimes it pauses by itself (pause-mark changes into play mark).
Any idea why? Too many viewers? (Tried in 3 consecutive days.)

lurch, 2015-03-17:
If the video is stuttering for you, try viewing the non-HD version by clicking on the blue 'HD' button at the lower right?

pickfire, 2015-04-03:
When I am playing it in Raspbian Wheezy's Epiphany browser, it started sluttering after 20 seconds and it crashes when I rewind. There isn't any option to use non-HD version.

And when I use Uzbl-browser in Raspbian Jessie, it only stutter once every 5 seconds, and I notice much more application is capable of playing video without a lot of stuttering in browser in Raspbian Jessie even thought I am not using the foundation's modified version of webkit.

LornaLynch, 2016-07-13:
Closing this as it appears to be superseded by newer versions of the guide.

closed 2016-07-13 by LornaLynch