raspberrypi/documentation #197: VNCs other than Tight are available

jamleed, 2015-03-18:
Here at RealVNC we've just ported our flagship VNC product to the Pi: http://www.realvnc.com/products/vnc/raspberrypi/

Reading the current doc you'd be forgiven for thinking Tight was the only VNC available: http://www.raspberrypi.org/documentation/remote-access/vnc/README.md.

We appreciate Tight are in the Debian repository system and RealVNC are a commercial organization now, but we actively promote free licenses, and there's the additional benefit that VNC can remote both console and virtual (Xvnc) sessions at the same time. Plus being easier to set up, in our opinion.

We'd like to edit the README to suggest that there are multiple flavours of VNC available, and namecheck ourselves as one of them.

Ruffio, 2017-01-07:
@jamleed this issue can be closed, right?

jamleed, 2017-01-09:
Yes, I think we're all done here, thanks a lot!

closed 2017-03-02 by JamesH65