raspberrypi/documentation #199: SD Card documentation isn't explicit about type of card to get

djmc, 2015-03-19:
I was recently gifted a Raspberry Pi 2, and needed to get an SD card for it. I went to the shop and checked last minute by doing a Google search and the `installation/sd-cards.md` page was the top result and implied the card should be an SD card. So I got one and was disappointed to discover when I got home that it should have been a MicroSD card.

It'd be nice if it explicitly stated the type of SD card to get for the model, for those who also stumble on that page looking for confirmation about what to get. It currently makes no mention about MicroSD cards.

lurch, 2015-03-20:
I agree that probably needs fixing, but I guess it's unfortunate that you found http://www.raspberrypi.org/documentation/installation/sd-cards.md before finding http://www.raspberrypi.org/products/raspberry-pi-2-model-b/ :-/

djmc, 2015-03-20:
I know, and I thought I was all read up on what the device needed. Oh well, I needed a better SD cars for my camera anyway :)

lurch, 2015-05-03:
Fixed with f663409943d04c8d3e1a9a14408f7a2192b27dc9

closed 2015-05-03 by lurch