raspberrypi/documentation #203: vnc init.d script refers to root user

AndrewFromMelbourne, 2015-03-27:
The [vnc documentation](https://github.com/raspberrypi/documentation/tree/master/remote-access/vnc) has an example init.d script to make the vncserver into a service. The script refers to the root user

    USER=root
    HOME=/root

which doesn't exist by default on Raspbian. Should this be

    USER=pi
    HOME=/home/pi

Discussion on the forum [here](http://www.raspberrypi.org/forums/viewtopic.php?f=91&t=105207&p=728271#p728271)

lurch, 2015-03-27:
Dunno much about VNC myself, but see also #194 and #133 (and #197)

AndrewFromMelbourne, 2015-03-27:
Yes, sorry @lurch should have had a better search through the existing issues! Do you want me to close this one?

lurch, 2015-03-27:
Feel free to close this issue, add a link to your forum thread to #194, and add a followup in your forum thread asking for feedback on #194 :-)

AndrewFromMelbourne, 2015-03-27:
Closing issue. It is a duplicate of [#194](https://github.com/raspberrypi/documentation/pull/194)

closed 2015-03-27 by AndrewFromMelbourne