raspberrypi/documentation #207: disable_audio_dither=1 is missing from config.txt docs

DougieLawson, 2015-04-01:
documentation/configuration/config-txt.md

Doesn't have ```disable_audio_dither=1``` documented.

lurch, 2015-04-01:
There's (unfortunately) lots of `config.txt` options that aren't documented yet. Pull Requests welcome... :)

DougieLawson, 2015-04-01:
If I knew what a "pull request" was and how to do one I'd do that.

There's another bit of missing docs.

pelwell, 2015-04-01:
You're just looking in the wrong place:

https://help.github.com/articles/fork-a-repo/
https://help.github.com/articles/using-pull-requests/

bennuttall, 2015-04-01:
> If I knew what a "pull request" was and how to do one I'd do that.

Click Edit when browsing a page of the docs on github. It'll give you your own version of the repo to edit. Commit changes and it'll give you the chance to send a pull request back to us to merge your changes in.

P33M, 2015-06-12:
Fixed

closed 2015-06-12 by P33M

kishorpise, 2020-05-21:
I just tried disable_audio_dither=1 on windows 10 iot core. It does not make any difference.

DougieLawson, 2020-05-21:
> I just tried disable_audio_dither=1 on windows 10 iot core. It does not make any difference.

No surprises there. The Windows IoT stuff is no longer supported by Microsoft and lacked that Linux specific function anyway. You'd be better off chosing any other operating system (or Linux distro) for your Raspberry Pi project.