raspberrypi/documentation #217: I think the page on passwordless SSH access has a mistake.

I-mess-up, 2015-05-09:
[Link to page](https://github.com/raspberrypi/documentation/blob/master/remote-access/ssh/passwordless.md)

I'm a beginner, so it's possible that I'm just confused, but I think there is a mistake on this page. In one line, it suggests using the code

    cat ~/.ssh/id_rsa.pub | ssh <USERNAME>@<IP-ADDRESS> 'cat >> .ssh/authorized_keys'

to copy the public key over to the pi. This didn't work for me, but once I added the path to the home directory (not sure if this is the right way to say this) it worked. My modified entry was 

    cat ~/.ssh/id_rsa.pub | ssh <USERNAME>@<IP-ADDRESS> 'cat >> ~/.ssh/authorized_keys'

Notice the extra "~/". 

Am I right that this is an issue or am I just a misguided user?

XECDesign, 2015-05-09:
When you pass a command like that, it runs in the home directory, so the two commands should behave exactly the same.
```
echo test | ssh xec-pi 'cat >> blah'
echo test | ssh xec-pi 'cat >> ~/blah2'
echo test | ssh xec-pi 'ls blah*'
  blah
  blah2
```

I-mess-up, 2015-05-09:
Alright. For me it didn't work but the change made it work. I can now connect to my Pi without a password. Is it ok if I leave the issue open for a bit just in case others want to chime in?

Another thing: is it possible I got some sort of error because I didn't have a directory named .ssh?

XECDesign, 2015-05-09:
Sure

Yeah, it's important to pay attention to error messages and paste them when reporting issues. However, adding a ~/ in that scenario would not have helped:
```
echo test | ssh serge-pi 'cat >> not-a-dir/blah'
  bash: not-a-dir/blah: No such file or directory
echo test | ssh serge-pi 'cat >> ~/not-a-dir/blah2'
  bash: /home/pi/not-a-dir/blah2: No such file or directory
```

I-mess-up, 2015-05-09:
Alright, I guess that I solved my issue by creating the directory then.  Thanks for explaining. I'll close the issue now.

closed 2015-05-09 by I-mess-up