raspberrypi/documentation #232: Recommendation to remove files after process is complete.

Nixpower, 2015-06-20:
I followed this: https://github.com/raspberrypi/documentation/blob/master/installation/installing-images/linux.md
and I thought it might be useful to append something that recommends you remove the files from your computer after completion becuase depending on the size of your SD card, the images could take an enormous amount of space on your disk, and you might not notice what's taking up your disk space like I did.

Many thanks,
nixpower

XECDesign, 2015-07-18:
That doesn't really seem to be within the scope of the document. It's up to the user what they do with the files on their computer and I would actually recommend keeping it in case the card needs to be re-flashed.

XECDesign, 2015-07-22:
Closing due to inactivity

closed 2015-07-22 by XECDesign