raspberrypi/documentation #234: Links to documentation on BCM2835 are gone

peterhgombos, 2015-06-23:
In the [page about the SOC](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bcm2835/README.md), there is a link to a Broadcom page with data on the processor, but this page has been removed from the Broadcom web page, and I can't find any alternative source on this.

Is there any current official source on the data of the SoC, i.e. not Wikipedia?

XECDesign, 2015-06-23:
Was there anything useful on that page? IIRC, it was mostly a 'contact broadcom for more information' page.

peterhgombos, 2015-06-23:
I have no idea. But I need an official site so I can cite the specifications of the processor for my master thesis, and I guess this page should contain it? I see Wikipedia also cites this missing page a lot.

XECDesign, 2015-06-23:
Checking through archive.org, it doesn't look like the page contained any real information https://web.archive.org/web/20150411220113/http://www.broadcom.com/products/BCM2835
Perhaps the peripherals manual and the relevant ARM documentation is the thing to cite?

XECDesign, 2015-06-23:
@bennuttall I guess the link should be removed?

bennuttall, 2015-06-24:
Yeah it looks like they don't have any info on the chips any more.

JamesH65, 2015-06-24:
I was unable to find any references to 2835 at all on the Brcm site. Not sure why. Perhaps ask Eben?

Doug-Wyman, 2015-06-24:
Is this of any help?

 

Never stop learning

Never stop Evolving

 

Doug Wyman

Doug-Wyman, 2015-06-24:
Mea Culpa, I don't see my attachment which was the 
BCM2835 ARM Peripherals PDF. I am new here so please forgive newby errors.

peterhgombos, 2015-06-25:
Thanks, @Doug-Wyman, but it seems that the document you sent is at a lower level than what I'm looking for. Not having read it all (I'm sorry, but hardware specs _is_ hard to read), I can't find any overview of the hardware available in it. And this was linked to on the same page, I think.

Doug-Wyman, 2015-06-26:
@peterhgombos  Ah well, it was worth a try.  I hope your master's thesis is accepted with honors.
And remember to keep on learning.

XECDesign, 2015-07-18:
Since the link is broken and there isn't anything to replace it with, I've just remove it.

closed 2015-07-18 by XECDesign