raspberrypi/documentation #235: Schematic for RPi 2 Model B

mtippett, 2015-06-27:
Seems to be missing from the collection.  

In particular, I am wanting to look at the layout to determine if the USB (power) port, can be connected to a PC and appear as a device.

ghollingworth, 2015-06-29:
No you cannot do this with a model B since the hub is not by-passable

lurch, 2015-06-29:
I guess this is possibly a duplicate of (or at least overlaps with) #224 #193 #188 #182

mtippett, 2015-06-29:
I'm fine with it not being possible to do what I was intending/hoping.  However, this issue itself is highlighting the gap in the RPi2 schematics.  If they are not released yet (or will not be released) the RPi schematic page should indicate as such.

ihartwig, 2015-08-25:
This would be really nice to see. It has been months. Can we work out the legalize? I am part of a university group that would like to develop low level ARM/embedded coursework around this board, and it is kind of hard to do that without the documentation.

JamesH65, 2015-08-25:
What specifically do you need that isn't covered by the pinouts and the
existing ARM peripherals documentation? The bare metal forum seems to be
active at the moment, so people do seem to be managing well without
official schematics.

ihartwig, 2015-08-25:
I am most interested in if these 2 blocks are the same between the Pi 1 B/B+ and Pi 2 B. I assume they haven't changed, but it is hard to tell when it is not documented.

[image]

[image]

bilderbuchi, 2016-06-20:
I have a similar problem. I want to figure out if there's a safe way to disable the red power led, as that even draws current when the RPi2 B I have is off. Without a schematic it's not easy to see if there will be knock-on effect of just cutting the trace leading to the LED.

@ebenupton will the schematics for the RPi2 be made available? If not, (while causing me an OSHW sadface) it would be great to put a clear statement concerning that on the [schematics page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md).

lurch, 2016-06-20:
...or you could simply disable the LED in software, like this: http://gpiozero.readthedocs.io/en/v1.2.0/recipes.html#controlling-the-pi-s-own-leds

Regarding schamatics, see also #347

bilderbuchi, 2016-06-20:
> ...or you could simply disable the LED in software, like this: http://gpiozero.readthedocs.io/en/v1.2.0/recipes.html#controlling-the-pi-s-own-leds

That doesn't help me when the RPi is off (as I noted in my comment above). The Pi is for a media server which seldomly runs, but if I leave the PSU connected the red LED is on 24/7. Maybe I'm a power efficiency nut, but I'd like to eliminate the pretty unnecessary power drain.

lurch, 2016-06-20:
Ahhh. When the Pi is 'off' it's not really off, the CPU is just held (AFAIK) in the HALT state. I suspect the SoC itself in the 'shutdown state' still probably draws more power than the little red LED? (and of course there'll also be power losses inside the PSU if you're leaving that plugged in, etc.)

> I'd like to eliminate the pretty unnecessary power drain.

Perhaps something like https://energenie4u.co.uk/catalogue/product/ENER002-1 would be a better option?

bennuttall, 2016-06-20:
Wouldn't allow you to turn the Pi off (ungracefully), but not back on?

lurch, 2016-06-20:
@bennuttall Huh? Which bit exactly are you referring to?
If you're talking about the remote-controlled socket, then you'd do a "normal shutdown" first, before hitting the button on the remote to kill the power to the Pi's PSU. And then when you hit the button (on the remote) again to restore power to the PSU, then the Pi would just boot up again as normal.

bennuttall, 2016-06-20:
Oh I see. I thought you were referring to turning off the Pi in software with the Energenie add-on.

lurch, 2016-06-20:
> I thought you were referring to turning off the Pi in software with the Energenie add-on.

Ahhh. No, I deliberately linked to just a 'plain' handheld-remote-control-only Energenie product ;-)

bilderbuchi, 2016-06-20:
I once measured the power draw downstream of the PSU, and a quick off-the-cuff calculation led me to believe that most of the shut down power draw came from the LED---unfortunately, I seem to have misplaced my notes regarding that, or I'd have specific figures.

My initial thought was to put together some self-holding relais-like circuit that opens as soon as the power draw falls to shut-off values, but I realised the components I'd need to buy would probably amount to more than the total savings from using the circuit, which doesn't make sense. For the same reason, I'd prefer not to buy additional products like remote-controlled power sockets etc. (but thanks for the hint!).

Then, I thought that by disabling the LED, I'd reduce the shut-off power draw to negligible values, and to wake it up I'd use a momentary switch I have lying around on the reset pins, so I don't have to build a circuit, and don't have to buy any components. This is where I am currently.

Maybe the best best after all is to just buy two micro-usb connectors and a small switch to splice between the PSU and the RPi, that way the Pi is guaranteed to be dead and the (low-standby-power) PSU does not have any load.

lurch, 2016-06-21:
> Maybe the best best after all is to just buy two micro-usb connectors and a small switch to splice between the PSU and the RPi, that way the Pi is guaranteed to be dead and the (low-standby-power) PSU does not have any load.

Something like https://www.pi-supply.com/product/pi-supply-raspberry-pi-power-switch/ ;-)

bilderbuchi, 2016-06-21:
Exactly, nice! OK, I think we should stop derailing this issue any further, thanks for any pointers/help!

closed 2016-11-23 by bennuttall

drsoubhi, 2018-01-18:
Hi everybody
I wish if anyone could send me the **datasheet** of **U16** "code 6343"  (SMPS) power IC of Raspberry Pi 2 as **PDF**
[image]

JamesH65, 2018-01-18:
AFAIK, we do not supply individual datasheet for components on the board. Sorry.

ebenupton, 2018-01-18:
You can find the Pi 2 PMIC datasheet here:

https://4donline.ihs.com/images/VipMasterIC/IC/ONSM/ONSMS37200/ONSMS37200-1.pdf

and the (less photosensitive) Richtek equivalent used on Pi 3 here:

http://www.richtek.com/assets/product_file/RT8088A/DS8088A-00.pdf

drsoubhi, 2018-01-18:
Dear Friends,
Thank you so much for your fast reply,

My issue is that my Raspberry Pi 2 U16 and after the use of low current SMPS (ie. 0.8A) +5V USB power supply seems to be damaged and overheated. My RPi 2 refused to restart. I tested the +5V and it's ok but only the two LEDs are lit! 
I would like to get out of the trouble since I'm living now in a very difficult place to reach. 
I will see the attachments and get back to you soon.

Kind Regards,

lurch, 2018-01-18:
https://www.raspberrypi.org/help/faqs/#power recommends a 1.8A PSU for the RPi 2.
You _might_ find that ordering a new Raspberry Pi is cheaper than trying to repair the broken one you have?

drsoubhi, 2018-01-18:
@lurch 

Perhaps you are right! (Buy a new RPi 2)
By the way I send it to my friend who needs to know more about such microcomputer. He told me that several mobile appliances use an equivalent chip of U16.
Also I need to investigate why insufficient power could suddenly damage the RPi 2!
I think that in RPi 2  the ** U16 = NCP6343 = DS8088A step down converter ** is adjusted to give a regulated 1.8V (3A)
I really couldn't realize where the damage is located!
**Perhaps** there is some other chip or semiconductor which drain more than the current limit and that's why the U16 started to overheat!. 
I need to have the __**complete schematic of the RPi 2 **__ to deeply study the issue!

JamesH65, 2018-01-18:
We do not supply a full schematic of the Pi devices, what we do release can be found here. https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md

karu2003, 2023-11-24:
Hi, I have it broken C90. What is its value? RPi 2 V1.1

commit eef7dcdb (2016-11-23): Add Pi 2 and Pi Zero schematics, close #224, close #235