raspberrypi/documentation #248: filezilla and spyware

gkreidl, 2015-08-25:
You should remove the link to filezilla-project.org and the recommendation to use it because it installs a lot of spyware on windows computers. ( remote access / sftp )

XECDesign, 2015-08-25:
Why? Does it use the sourceforge malware installer?

gkreidl, 2015-08-26:
See this report on the RPi Forum:
https://www.raspberrypi.org/forums/viewtopic.php?f=62&t=118996&p=807374#p807400

I didn't test it myself, didn't want to infect my last remaining windows computer with anything.

XECDesign, 2015-08-26:
I am unable to confirm that claim and can see that there are quite a few contradictions there. The installer only contains filezilla and doesn't ask you to accept anything other than the GPL. The page doesn't link to CNET and you get the same installer whether or not you click the recommended link or not. "show additional download options" shows downloads for other operating systems, in addition to the same installer.

XECDesign, 2015-08-26:
Okay, just confirmed this. However, it's not spyware, just some PUPs, which the user has to agree to before they are installed. That's pretty standard in the windows world and there is a clear warning on the download page.

BrianMikinski, 2019-03-17:
# This needs to be opened again and investigated

Downloading from the windows msi from [https://filezilla-project.org/](https://filezilla-project.org/ ) and casually clicking through the installer placed

**Unwaders.C!ml** and **Prifou.A!cl** into some temp app data folders which was blocked by Windows Defender as a severe threat. Upon further investigation this these files look like [very dangerous trojan horses](https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Misleading:Win32/Unwaders.B!ml&ThreatID=242866)

Please investigate. It doesn't seem like FileZilla should be recommended if the default installer from the developers website pushes the user towards viruses and trojan horses

See the following links for more recent information - 

- [(https://forum.filezilla-project.org/viewtopic.php?t=48441](https://forum.filezilla-project.org/viewtopic.php?t=48441)
- [https://www.reddit.com/r/netsec/comments/8t4xrl/filezilla_malware/](https://www.reddit.com/r/netsec/comments/8t4xrl/filezilla_malware/)

XECDesign, 2019-03-17:
Maybe the documentation should recommend winscp for windows and filezilla for Linux? Whether or not it's a false positive, as the author claims, it's still annoying to have software that actively tries to trick you into installing useless junk.

JamesH65, 2019-03-18:
A quick Google and Filezilla does seem to be doing some shady stuff. I'l take a look at the docs.

JamesH65, 2019-03-18:
Would this change or similar be appropriate?

https://github.com/raspberrypi/documentation/pull/1104

BrianMikinski, 2019-03-18:
@JamesH65 looks good to me. I'm not sure what other cross plat file transfer applications exist like FileZilla but any tool that is easy to use and safe is fine with me.

JamesH65, 2019-09-18:
#1104 merged, so closing.

closed 2019-09-18 by JamesH65