raspberrypi/documentation #249: Audio URL for la.mp3 changed currently not found/404

jaypl, 2015-09-05:
In the doc at
[audio/README.md](https://github.com/raspberrypi/documentation/blob/master/usage/audio/README.md) :sound: 

The listed example.mp3 download link is broken (http://goo.gl/MOXGX3)

The new location is:
https://github.com/raspberrypilearning/burping-jelly-baby/blob/master/sounds/la.mp3

Perhaps a new link as bit.ly shortcut (allows updates) would be best

LornaLynch, 2015-09-07:
I've updated the link, but I've kept it as a Google-shortened link in case there are tracking/analytics reasons for this.

bennuttall, 2015-09-08:
The sound file ought to be removed from the jelly baby resource as it's no longer used.

Perhaps we should just do what we do in the resource and copy a file from the Sonic Pi samples folder.

FYI there's no tracking on the shortened URLs - just much shorter to type.

bennuttall, 2015-09-08:
Also the change did not fix the 404 as it's now linking to the GitHub webpage showing the file - not the location of the file itself - I'll fix it.

LornaLynch, 2015-09-09:
Ack - sorry @bennuttall - do you need me to do anything or are you on it?

commit 7f70a187 (2015-09-10): Fix broken mp3 URL, close #249
closed 2015-09-10 by bennuttall