raspberrypi/documentation #254: DSI Display Orientation

jsavage, 2015-09-22:
The readme for the DSI display does not cover the key facts about the intended orientation of the display.  It mentions 130degree viewing angle but no reference point or orientation is mentioned.   A simple diagram would help. 
https://github.com/raspberrypi/documentation/blob/master/hardware/display/Readme.md

The mechanical drawing does not clarify this point because it shows the viewing direction as being vertically upwards.  Please could that also be clarified.

JamesH65, 2017-03-02:
Corrected link 
https://github.com/raspberrypi/documentation/blob/master/hardware/display/README.md

However, not sure what you are requesting here. Can you clarify what it is you want?

JamesH65, 2017-03-15:
Closing as no response from OP. Feel free to reopen if necessary

bennuttall, 2017-03-15:
@JamesH65 I think this is valid. The display is better viewed one way up than the other - and needs to e set in config.txt:

> Help, my screen is upside-down!
> 
> Note: An update has been pushed to Raspbian to flip the screen ( rotate it by 180 degrees ) for a better desktop viewing angle. This makes it upside-down in our stand and the official Pi stand, so you'll need to change a setting to flip it back.
> 
> To do this, open `/boot/config.txt` in your favourite editor and add the line:
>
> `lcd_rotate=2`
> 
> This will rotate both the LCD and the touch coordinates back to the right rotation for our display stand.
>
> Don't use the documented `display_rotate`, it performs a performance expensive rotation of the screen and does not rotate the touch input.

From http://forums.pimoroni.com/t/official-7-raspberry-pi-touch-screen-faq/959

JamesH65, 2017-03-15:
@bennuttall OK, happy to tech proof read a PR!

JamesH65, 2017-05-16:
Fixed by #643

closed 2017-05-16 by JamesH65