raspberrypi/documentation #255: Official Touchscreen LCD FAQ, Instructions, ETC

Gadgetoid, 2015-09-22:
This is regarding the Readme.md, and possible other documents which would reside here: https://www.raspberrypi.org/documentation/hardware/display/Readme.md

I've been collating a list of FAQs over at the Pim forums in an effort to get my head above the tide of questions and support around the official touchscreen: http://forums.pimoroni.com/t/official-7-raspberry-pi-touch-screen-faq/959

Specific topics from this which I think could be ported into the documentation include:

* Instructions for update/upgrade, but avoiding `rpi-update` or at least only with disclaimers
* Standard first steps for debugging if;
    * The touchscreen doesn't work - check ribbon, update
    * The LCD is white - check LCD ribbon
    * The LCD is black - check ribbon to Pi
* Clarification of HDMI output support, and that dual display X is not currently supported
* Clarification that the ability to turn on/off in software and control brightness is coming
* Steps for using `lcd_rotate` and clarification that there isn't really a *wrong* orientation
* Clarification that a 2A supply is pretty much necessary
    * Display uses between 450/470mA and tops out at ~925mA when coupled with a Pi 2
    * Rainbow square means your supply isn't good enough, or that your GPIO/USB cable is flaky/causing a voltage drop
    * Split USB cable might be the best bet
* Choices for virtual keybowrds: Florence/Matchbox
* Details of reverse mount configuration highlighted by Clive Beale, that it works but basically makes GPIO inaccessible because a ribbon cable wont fit.
* List of supported operating systems
* List of supported Pi models ( model B, A currently not working )

nielsswinkels, 2015-11-27:
Would be nice to add instructions on how to calibrate the touchscreen.

Gadgetoid, 2015-11-27:
As far as I'm aware, there's no calibration process. If you're seeing your touches misaligned with the screen it may possibly be an error in your config. I can't remember exactly what the fix was.

bennuttall, 2016-01-28:
Fixed in 122708bac490ee8468624c779b86620cec803edc and others

closed 2016-01-28 by bennuttall