raspberrypi/documentation #266: Local kernel build speedup on Rpi2

aahlborg, 2015-10-10:
The page Kernel Building page (https://www.raspberrypi.org/documentation/linux/kernel/building.md) mentions that building the kernel locally on the raspberry pi takes a lot longer than cross compilig. However, in the case of a rpi2, the build process can be accelerated by a factor 4 (from ~400 minutes to ~100 minutes) by utilizing all four cores. All that's needed is adding the flag -j4 to the make command.

The documentation should mention this simple trick as an alternative to setting up cross compiling on a remote host, which requires much more work.

pelwell, 2015-10-12:
I think the subsequent note covers this adequately:
> Note: To speed up compilation on multiprocessor systems, and get some improvement on single processor ones, use -j n where n is number of processors * 1.5. Alternatively, feel free to experiment and see what works!

waveform80, 2015-10-12:
That note is under the "cross-compiling" section which I could forgive someone for skipping if all they're interested in is a local build. It might be worth repeating that admonition at the end of the local installation section.

pelwell, 2015-10-12:
Fair enough. I've added a note (more suggestion than admonition) within in the "Local building" section.

waveform80, 2015-10-12:
Sorry, I meant admonition in a more general sense (notes, hints, info callouts, warnings, and what not are all referred to as "admonitions" in reST so I tend to think of the word as describing that class of things, completely forgetting that it's got a rather different connotation in general English! :)

aahlborg, 2015-10-12:
I'm happy with that note.

closed 2015-10-12 by aahlborg