raspberrypi/documentation #273: Update Documentation for VNC using systemd

jrobinson-uk, 2015-10-22:
To make VNC work under Jessie (systemd) the following system.d unit file works

/etc/systemd/system/vncserver@:1.service

[Unit]
Description=Remote desktop service (VNC)
After=rsyslog.service,network.target

[Service]
Type=simple
RemainAfterExit=yes
ExecStartPre=/bin/sh -c '/usr/bin/vncserver -kill %i > /dev/null 2>&1 || :'
ExecStart=/sbin/runuser -l pi -c "/usr/bin/vncserver -geometry 1024x768 %i"
ExecStop=/sbin/runuser -l pi -c "/usr/bin/vncserver -kill %i"

[Install]
WantedBy=multi-user.target

The service is enable on boot with

"sudo systemctl enable vncserver@:1.service"

Ruffio, 2017-01-07:
@jrobinson-uk can this issue be closed?

JamesH65, 2017-03-02:
AFAIK, VNC now works out of the box on the latest Rasbian Jessie, so closing. Please reopen if there is still an issue.

closed 2017-03-02 by JamesH65