raspberrypi/documentation #278: Compute Module IO board and the Official 7in Display

adam-lee, 2015-10-28:
I am submitting this issue to let maintainers know that I got the 7in display working on the IO board with the help of 6b79 [1], and I will be sending pull requests. 

Please let me know if this isn't the place to do it!

[1]https://www.raspberrypi.org/forums/viewtopic.php?p=835160#p835160

bennuttall, 2016-01-28:
Since #279 and #302 are resolved, can this be closed?

closed 2016-01-28 by bennuttall

adam-lee, 2016-01-28:
yes - thx

piwizards, 2017-02-20:
Purpose: - Just to use Official 7-inch screen as display

Using the following hardware:-

Compute Module 3 Lite
Compute module IO board V3.0
Official 7-inch touchscreen display
22 way to 15 way DSI display adapter
Official 5v power adapter
Software Info:-
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian

Followed the following process to connect:-

After running vcgencmd version, received the following output:-
Dec 9 2016 15:11:26
Copyright (c) 2012 Broadcom
version 2e557d8dac70add28597c3b449cb52c34588d818 (clean) (release)

Connected the display to the DISP1 port on the Compute Module IO Board through the 22W to 15W display adaptor.
Connected these pins together with jumper wires:
GPIO0 - CD1_SDA
GPIO1 - CD1_SCL
Executed the following command on the Compute Module 3 Lite:-

sudo wget https://goo.gl/Ah6XD5 -O /boot/dt-blob.bin

Rebooted for the dt-blob.bin file to be read, but the screen is blank.
Also supplied 5V extra power supply to PWR IN port of the touch screen.
But still, the screen is blank.
[image]

pelwell, 2017-02-20:
This is _still_ the documentation repo - you will not get application support here.

6by9, 2017-02-20:
But seeing as you posted
https://github.com/raspberrypi/documentation/issues/498
https://github.com/raspberrypi/documentation/pull/509