raspberrypi/documentation #282: Raspberry Pi Zero - schematics and usb on pins

saper-2, 2015-11-26:
Hi,

There is no schematics, only mech drawing :cry:  .
I beg you, publish it, or at least give description of test points, because I would like to hook an usb hub (and maybe even LAN9514) directly, not by big sticking out microUSB connetor :disappointed: , and for powering Pi0 also I don't like the idea of "big, fat microUSB plug sticking out of board...." (board that is not much longer than usb plug...)

lurch, 2015-12-09:
It's obviously "at your own risk" but have a look at http://hackaday.com/2015/11/28/first-raspberry-pi-zero-hack-piggy-back-wifi/

And the Pi Zero is HAT-compatible, which means it's possible to back-power it using the GPIO header, as described on https://github.com/raspberrypi/hats

mtrudel, 2016-09-02:
Any update on Pi Zero schematics?

lurch, 2016-09-02:
I suspect the answer to this is the same as in #347

mtrudel, 2016-09-02:
In my particular case I'm wondering about the specifics of back-powering the Pi via the GPIO header. I know that there's a couple of statements on the matter in the [HaT documentation](https://github.com/raspberrypi/hats/blob/master/designguide.md), but a lot of people (myself included) would feel better (and learn more) if we could interpret this guidance in the context of an actual schematic.

I don't think anyone is asking for schematics all the way into the SoC (I'm certainly not), but it would help lots of people to have more insight into external (GPIO and RUN header, mostly) interfaces without having to resort to the hearsay and speculation that exists on Stack Overflow et al in the absence of hard data from the Foundation.

lurch, 2016-09-02:
It's still not clear exactly what info you're after, but if you look at e.g. https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/Raspberry-Pi-Rev-2.1-Model-AB-Schematics.pdf you can see that most of the pins on the GPIO header just connect directly to the SoC. AFAIK the RUN header on the Pi Zero is identical to the RESET header on the Model B.
And while it's not "official", some of the links in this thread might be useful https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=86721

mtrudel, 2016-09-02:
Thanks, @lurch; I'd not seen that last thread. In an abstract sense, those sorts of forum posts are exactly what I'm talking about. They're helpful, but they're not canonical documentation.

In my particular case I'm working with the Pi Zero which has (to my knowledge) no published schematics, simplified or otherwise. I realize the implications of the GPIO header's 5V line being after protection in the other Pi versions and everything that that entails from a back powering perspective, but without any schematics for the Zero I'm working on a combination of reverse engineering, community commentary and best guesses to figure out the best way forward to back power a Pi Zero. 

To be honest, I realize that the community (and myself) are asking you guys for a lot, considering that you've built nothing short of a miracle in the form of a $5 computer, and that this kind of thing is very much an edge case from your perspective. If your position is to not supply schematics, that's completely understandable. But on the other hand, it would be nice to have an officially articulated position on the matter one way or the other.

Ali-Chen, 2016-09-11:
Reason for request for "better" schematics is that the abbreviated variant does not have circuitry around the downstream USB connector. There are concerns that the VBUS power on USB is inadequate and does not meet USB specifications, due to use of under-rated poly-switches. Older Pi shows miniSNDC014 fuses rated at 140mA with cut-off at 280mA, which is vastly inadequate to power external HDD enclosures.

lurch, 2016-09-11:
The easiest way to avoid under-current on USB is to simply use a powered USB hub.

Ali-Chen, 2016-09-11:
A powered hub costs money, plus inconvenience to have another wall adapter and ugliness of extra wires. The best way however is to comply with USB specifications, which may not cost anything other than a will. Are you saying that Pi-3B still uses 140mA poly-switches? What about compliance to droop test?

LizUpton, 2016-09-11:
As we've said before, we will not be releasing full schematics. Sorry to those of you this disappoints. This being the case, and @lurch having already provided useful links, I'm closing this issue.

Of course, Pi Zero (and all our other models) are fully compliant with all industry standards - if they weren't, we wouldn't be able to sell them. (And we wouldn't want to; any organisation skimping on compliance is not playing fair with other organisations in its field, or with customers.) As for the poly-switches: wow, that's a blast from the past! We haven't used those in years (this goes several models back).

closed 2016-09-11 by LizUpton

E3V3A, 2016-09-14:
@lurch @LizUpton 
> As we've said before, we will not be releasing full schematics. 

That is a real shame. It is clear from your old schematics for the RP1B that the CPU USB pins does indeed go to one of the  USB ports. If this schematic is enough similar for the **RPi3B**, as you always claim, then that means that the *LAN9514* 4x USB HUB chip can be circumvented completely and `OTG` should be possible by cutting the CPU SOC pin `USB_OTGID` and pull it high, which would be an easy and great hack. I mentioned and showed this in your forum [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1037540#p1037540). 

So yes, please post the equivalent few USB signal lines for the RPi3B. 

Specifically what we want to know is:

- What pin is the CPU SOC  `USB_OTGID`?
- Where are the equivalent 0-Ohm R28/29?
- Where are the SOC `USB_DP/DM` pins and signals going?
- To what USB port are they routed?

Ali-Chen, 2016-09-14:
I am not sure I understand. Yes, you likely can re-route everything and ID
pin to make the port as dual-role compatible. But who is going to write the
device side of software on Pi3 side to control the ID switch, VBUS switch
(if any), and write support for all endpoints and overall USB response
protocol? As I understand, the Broadcom uses SNPS sloppyware as USB device
controller, and the software task can be quite challenging. Do you plan to
write it?

lurch, 2016-09-15:
In case I've given anyone the wrong impression, I'd just like to say that I'm not a Raspberry Pi employee, I'm just an enthusiastic community member.

> which would be an easy and great hack

Have you actually *seen* the size of the components, and how close they're placed together, on an RPi 3B board? Attempting **any** kind of modification is hardly something I'd call 'easy'!

> Where are the equivalent 0-Ohm R28/29?

IIRC those were only on the original boards, because the same PCB was used for both the Model A (R28 & R29 fitted, LAN9512 not fitted) and Model B boards, (LAN9512 fitted, R28 & R29 not fitted).
As the Pi3B board *always* has a LAN9514 fitted, I wouldn't expect the equivalent 0-Ohm resistors to be there.

> Where are the SOC USB_DP/DM pins and signals going?

Again, I'm only guessing, but I'd expect them to go directly from the BCM2837 to the LAN9514 (perhaps through some vias on the way).

> But who is going to write the device side of software

That's already been done, and is something the PiZero supports, see e.g. http://blog.gbaman.info/?p=699 and https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/other-modules

E3V3A, 2016-09-19:
@lurch 

>    which would be an easy and great hack

> > Have you actually seen the size of the components, and how close they're placed together, on an RPi 3B board? Attempting any kind of modification is hardly something I'd call 'easy'!

I meant a kernel hack in comparison to a PCB SMD hack. 

The old crappy schematics of the RPi1 is a bit confusing without the PCB silkscreen layout. What does it mean that R28/29 are "fitted"? Usually a zero Ohm R are just presented as a solder patch. Does this even exist on the RPi3? Probably not. In addition, the `USB_DP/DM` lines most likely go **only** to the LAN9414. But this remain inconclusive until someone either gives us the proper schematics or X-rays the PCB.

lurch, 2016-09-19:
> In addition, the USB_DP/DM lines most likely go only to the LAN9514. But this remain inconclusive until someone either gives us the proper schematics or X-rays the PCB.

True. But all the RPi devices are designed to be as cheap as possible, so I'd be very surprised if it wasn't just as simple as that.

If the USB OTG mode supported by the PiZero isn't fast enough for you, perhaps you'll just have to wait for the BCM2837-based CM3. \*shrug\*

gtsiros, 2016-10-22:
Amidst all the cuteness and feeling of openness, this comes as a slap in the face of everyone who contributed to the rpi

JamesH65, 2016-10-25:
The Raspberrypi is not and never has between an open source project. A lot of effort goes into making it as OS as possible but there limits. As a long term contributor in various capacities I understand that, there is no slap in the face here.

AymenLA, 2016-10-25:
Since it Is BRCM that sources the schematics and the main chip, I don't think there will be any official schematic availble. If it was TI/ MCHIP/ ST that would have been possible.

saper-2, 2016-10-25:
I think this issue can be closed. brcm _(they don't deserve for uppercase)_ is [there goes a long list of censored synonymous terms] and this is period. I don't think the brcm made schematics for RPi's, they only source chips , dev docs & tools for the Foundation.

Well, I like to check myself what is inside things, I don't like "black boxes". I like to analyze, and learn about part used in circuit. I'm cracking open almost everything I buy :smile: .

It was worth asking, but answer was nothing new. Well I'll try with next Pi's too :stuck_out_tongue: 

!close :smile:

lurch, 2016-10-25:
Presumably all the Open Source zealots / fanatics in this thread have been busy poring over the CM3 schematics? ;-)

P.S. Hello again JamesH :)

> I think this issue can be closed.

It already is ;-P

saper-2, 2016-10-25:
Wait, Compute Module based on Pi3 cpu ? :open_mouth: I missed this news.... So I've to secure some founds :smile: (now waiting for kit :) ) But I have to look for other supplies for CM3 because all suppliers in my country want over 140USD (~100..120GBP) for **CM1** - I don't even want to think how much those swindlers will want for CM3...

Closed, oh - missed this too :expressionless:

lurch, 2016-10-25:
> Wait, Compute Module based on Pi3 cpu ? :open_mouth: I missed this news....

It's not been released yet, but https://www.raspberrypi.org/blog/compute-module-nec-display-near-you/ says it should be available around the end of the year. And the schematics are already available https://www.raspberrypi.org/documentation/hardware/computemodule/schematics.md

Regarding the price you mentioned - that might be for a CM dev kit https://www.raspberrypi.org/products/compute-module-development-kit/ rather than just a CM module https://www.raspberrypi.org/products/compute-module/ on its own?

saper-2, 2016-11-13:
Well, I won't pay 140USD++ - that's crime. 
I found that farnell (& elements14) add at least about 20% to price - farnell have CM1 kit for ~105USD, while RS have CM1 kit for ~75USD...
Anyway, I'm waiting for CM3 to being publicly available :grin: then I'll see what to do next :)
!eot :)