raspberrypi/documentation #285: Unable to copy keys for passwordless login using this method

jsavage, 2015-11-30:
Hi,
This is an issue with Passwordless.md
I am not sure how to fix this but when I followed the suggested method for copying the content of the local file ~/.ssh/id_rsa.pub to remote file .ssh/authorized_keys I encountered an error.  The actual command I was using was as follows. The only thing I have changed is the IP address and the port number:
$ cat ~/.ssh/id_rsa.pub | ssh -p26 pi@xx.xx.xx.xx 'cat >> .ssh/authorized_keys'

bash: .ssh/authorized_keys: No such file or directory

On investigation I found that the remote Raspberry Pi did not have an existing folder .ssh or the authorized_keys file.  This is Pi  has a fairly new install of Raspbian Jessie by the way.  I am fairly sure I have never deleted the .ssh directory so I suspect the move to Jessie may be the root cause.

I resolved this by creating the folder and file manually and then copying and pasting the content over manually as suggested in http://raspi.tv/2012/how-to-set-up-keys-and-disable-password-login-for-ssh-on-your-raspberry-pi.  However, a command line solution would be preferable.

If I work out a solution I will send a pull request to fix this.

James

lurch, 2015-11-30:
Looks like this is probably the same problem that was also described in #217 ?

Can you confirm if it works if you change the command to
`cat ~/.ssh/id_rsa.pub | ssh <USERNAME>@<IP-ADDRESS> 'mkdir -p .ssh; cat >> .ssh/authorized_keys'`
 (all on one line) ?

jsavage, 2015-11-30:
Yes - it seems to be a similar issue.
I have submitted a Pull request which includes a similar one-liner and some words of explanation but my suggestion was for that to be used only if this error arises not as a replacement for the existing command. The mkdir does not need the -p  in those circumstances.
I tested your soln by deleting my .ssh directory and it worked for me.  What I can't be sure about is whether .ssh and authorized _keys are supposed to be there in Raspbian by default.

Ruffio, 2017-01-07:
@jsavage can this issue be closed?

JamesH65, 2017-03-02:
Fix has been merged so closing. Any furthur issues, please submit a new Issue/PR.

closed 2017-03-02 by JamesH65