raspberrypi/documentation #294: VNC Instructions require some modifications

seancull, 2016-01-03:
When following the instruction ( https://www.raspberrypi.org/documentation/remote-access/vnc/ ) on a vanilla install of RASPBIAN JESSIE I had the following 2 issues,

1) If the default setting of auto login is enabled then the session : 0 cannot be used and the error message below is created.
Warning: raspberrypi:0 is taken because of /tmp/.X0-lock
Remove this file if there is no X server raspberrypi:0
A VNC server is already running as :0

see http://stackoverflow.com/a/34577704/1167715

2) When trying to automate the vnc startup using the command line 
update-rc.d /etc/init.d/vncboot defaults

I get the error 
initscript does not exist: /etc/init.d//etc/init.d/vncboot

Using the command line 
update-rc.d vncboot defaults 
works OK

seancull, 2016-01-03:
sorry - ignore item 2 - it is covered in the instructions

lurch, 2016-01-29:
Looking at the current content of that page (it had some recent updates), I guess this issue can be closed now?

closed 2016-07-13 by LornaLynch