raspberrypi/documentation #299: linux - filesystem - backup.md : backing up Home directory does not work

DeeJay, 2016-01-21:
As documented:
cd /home/
tar czf pi_home.tar.gz pi

Result:
pi@raspberrypi ~ $ cd /home/
pi@raspberrypi /home $ tar czf pi_home.tar.gz pi
tar (child): pi_home.tar.gz: Cannot open: Permission denied
tar (child): Error is not recoverable: exiting now
pi@raspberrypi /home $

Clearly user pi does not have write permission into /home. Can be fixed by invoking tar with sudo.

DeeJay, 2016-01-21:
A note about how to restore from this backup would also be beneficial.

bennuttall, 2016-01-28:
Good point! PR welcome...

lurch, 2016-01-29:
I guess "how to restore from this backup" would depend on whether you only want to overwrite particular files with older versions (in which case a graphical archive utility would be your best bet) - or whether you actually want to restore *everything* from the archive, potentially overwriting (or losing) any recent work you've done... ?

JamesH65, 2017-03-02:
Currently waiting for confirmation to merge the linked PR.

LornaLynch, 2017-03-02:
I've edited and merged the PR. Closing this.

closed 2017-03-02 by LornaLynch