raspberrypi/documentation #306: Once again we've reached a point where https://github.com/raspberrypi/documentation/blob/*/configuration/config-txt.md#configtxt doesn't match reality

DougieLawson, 2016-01-31:
There's been a lot of new parameters added. Some are documented in /boot/overlays/README some are missing altogether.

If we take 
```
        act_led_trigger          Choose which activity the LED tracks.
                                 Use "heartbeat" for a nice load indicator.
                                 (default "mmc")

        act_led_activelow        Set to "on" to invert the sense of the LED
                                 (default "off")

        act_led_gpio             Set which GPIO to use for the activity LED
                                 (in case you want to connect it to an external
                                 device)
                                 (default "16" on a non-Plus board, "47" on a
                                 Plus or Pi 2)

        pwr_led_trigger
        pwr_led_activelow
        pwr_led_gpio
                                 As for act_led_*, but using the PWR LED.
                                 Not available on Model A/B boards.
```
from README as examples there's not enough documentation there to use those parms. For example what value can I set for pwr_led_trigger? What's the default value? Can I swap the function of the act & pwr LEDs?

Can someone add some more of the missing pieces so that the website has more complete docs for the GPU code parameters which are currently exceedingly opaque. If that's not possible can someone expand the README to fully document any parms you'd expect end-users to tweak.

JamesH65, 2017-03-02:
Hi Dougie, is this still an issue?. I've checked the docs and they look fairly comprehensive, but there may still be bits missing.

James

DougieLawson, 2017-03-02:
What I was hoping for was for someone to go through the whole bootcode.bin and document every possible parameter (that hasn't been retired or deprecated). So we can have an absolutely definitive reference of every parm and every keyword value that's applicable.  You probably never use this doc item because you're exceedingly clever and have been working with this stuff every day for years (like me). If the source was open I'd have done that and created a pull request (who said I'm not anal like that). No - don't open the source, it will be a can of worms with a frosting of stinky stuff.

Look at it from the point of view of the brand new Raspberry Pi Zero with WiFi user who's never met anything like /boot/config.txt ever before (there's nowt like that that isn't hidden on Windows since we left W95 behind). The parms are daunting. Most of them aren't needed (most of my 14 RPis have three lines in config.txt).  There's some conflicting stuff on the elinux.org wiki as well. We need to be able to point to the RaspberryPi.org pages as say "This is the only reference you need." (plus or minus /boot/overlays/README).

Maybe I've worked with IBM Mainframes too long as they have everying documented as their documentation policy and accept doc changes when you find an undocumented parm or one where code and docs don't match. They'll even take bug fixes if the docs are badly enough broken that they crash your system. It's been like that for 35 years (I used to rip out the Reader's Comment Forms at the back of IBM paper books and mail them in). I even have a special place in the inbox for IBM's IMS Transaction and Database Management system (not /dev/null) with a fast route to their documentation dept.

It becomes a work in progress but doing a review for each change must be a bunch easier than trying to fix something that's got stale. 

BTW, while you're fixing things you should also document which parms set OTP bits/bytes and are sticky after first use (like codec keys and USB/PXE boot). Neat trick doing that - it makes life simpler.

JamesH65, 2017-03-02:
Thanks Dougie, I see what we can do - all sounds like the right approach. Once it's done it should, as you say, be fairly easy to maintain. Cannot promise timeline as a bigger job, but it's now on my radar/assigned.

JamesH65, 2017-03-08:
First stage of this, splitting up the huge config.txt file in to subfiles, has been merged. Not real additional content although all files have now been copy edited.

JamesH65, 2017-03-15:
#569 has updated the overclocking page to current data.

Still need to check the other pages for accuracy. Given the huge number of issues found on the overclocking page that may take a while...

JamesH65, 2017-05-16:
I've started on this - might be a while...

JamesH65, 2017-05-18:
The PR has been merged, so closing.

closed 2017-05-18 by JamesH65