raspberrypi/documentation #308: linux commands displayed in uppercase due to style sheet

sebe, 2016-02-05:
The mark up for this page is fine
https://www.raspberrypi.org/documentation/linux/usage/commands.md

but the style sheet, style.css under wp-content/theme/mind-control has text-transform: uppercase;
These command headings are h3, maybe you could change the tag or the css.

h1, h2, h3, h4, h5, h6 {
    font-weight: 400;
    text-transform: uppercase;
}

lurch, 2016-02-05:
The same problem also affects https://www.raspberrypi.org/documentation/configuration/config-txt.md

bennuttall, 2016-02-05:
If it's enough of an issue I could make all headers in documentation not uppercase. I'm not sure it is...

lurch, 2016-02-05:
I'm not a markdown expert, but would there be any way to 'tag / mark' specific headers in the documentation that *shouldn't* be auto-converted to upper-case? (i.e. so that they'd get rendered by a slightly different CSS rule, while allowing all non-marked headers to be auto-upper-case as usual)

bennuttall, 2016-02-05:
There's obviously nothing in the markdown spec (nor is there a markdown spec at all, right?) for styling elements, as that's not what it's for. It's possible to throw my own convention in like:

```
## {} config.txt
```

translates to:

```html
<h2 class="not-uppercase">config.txt</h2>
```

but the more stuff like that you do, the more potential for errors there are. See some of the weird issues we get on the resources when relative links aren't working...

I'd rather make a rule for the whole documentation than permit individual override.

Ruffio, 2017-01-07:
@sebe can this issue be closed?

JamesH65, 2017-03-02:
@bennuttall Is this still an issue, or can I close it?

lurch, 2017-03-02:
It does seem odd, having all the commands listed in uppercase, when that's not how they're typed in (and obviously trying to do so would result in command-not-found errors.
If you don't want to modify the header-styles, I guess another way around this issue might be to show a usage example (with the command correctly displayed in lowercase) underneath each (uppercase) header?

bennuttall, 2017-03-02:
I don't think it's an issue - it's styling the headers like the rest of the website. They're clearly titles, not examples. The examples are shown as they should be.

Do you have an opinion, @JamesH65?

JamesH65, 2017-03-02:
It is a bit odd, since not all commands have examples, but I wonder is a line right at the top to explain all/most linux commands are in lower case and these are just captialised a good compromise.

LornaLynch, 2017-03-02:
If this would be helpful I could edit the text to ensure the command is correctly displayed in each section of explanatory text. Most of the explanations have this anyway, so it shouldn't be a big deal to reword the others a touch.

LornaLynch, 2017-03-06:
@JamesH65 give me a shout if you are happy for me to go ahead and do this.

JamesH65, 2017-03-06:
I think its a good idea -the captialised headers do look odd when they are
clearly used in lower case in action.

LornaLynch, 2017-03-06:
Great - thanks! Will go ahead and give you a shout when I'm done.

LornaLynch, 2017-03-06:
Done: hopefully this should be a lot clearer now.

closed 2017-03-06 by LornaLynch

lurch, 2017-03-06:
Looks much better now :+1:

LornaLynch, 2017-03-06:
Thanks @lurch!