raspberrypi/documentation #317: Mac Connection Documentation is incorrect

deneb711, 2016-02-21:
The instructions state that if display :1 was started on the pi, then the command should be vnc://raspberry.local:1

That seems to be incorrect based on my understanding of how the command works and based on my own experience. The correct command should be:

vnc://rapsberry.local:5901 for display :1

If the display is :2, the command should be 

vnc://raspberry.local:5902

Otherwise the connection is refused by the Pi at least it is for me.

lurch, 2016-02-21:
This was changed fairly recently in #303 - perhaps the correct parameters to use depend on exactly which VNC client software you're using? (and/or version of MacOS?)
Any comments from @teschlg ?

deneb711, 2016-02-21:
I'm using the built in vnc capability in OS X, i.e. "Connect to Server..." from the Finder>Go Menu. 

I also downloaded VNCViewer and ran into exactly the same problem. If you specify :1 or :01 you will get your connection refused. If you specify :5901 you'll be able to connect. In fact, the screenshot that's currently partly of the instructions shows :5901 and not :1 or :01 so it may well be that an incorrect change was recently made.

I am on OS X 10.11.3 but the VNC:// functionality in the finder has been unchanged for several years, so I don't think this is a version issue. VNC uses port 5900, and the last digit should reference the display created by VNC Server, so it makes sense that the correct syntax is :590X.

teschlg, 2016-02-21:
Yes, I can confirm this, the port must be “:5901" and not “:1”. I was originally using display :0 in which case there is no need to specify a port at all.
Sorry.

JaredBoone, 2016-05-29:
I can confirm that this issue is also present w.r.t. OS X 10.10.5. Pull request #318 appears to correctly address the issue.

closed 2016-07-13 by LornaLynch