raspberrypi/documentation #321: Raspberry Pi 3 B Schematics

jorgegarciadev, 2016-03-03:
Greetings,
Any plan on releasing the schematics for the Raspberry Pi 3 B?

bennuttall, 2016-03-04:
Yes - they'll be added soon.

wilfm, 2016-03-13:
Could you also please make sure there are the mechanical drawings as well?

marc937, 2016-03-31:
Any ideas when the schematics will be available? Thanks!

closed 2016-04-04 by bennuttall

awneil, 2016-04-12:
Why are there only "reduced" schematics?

Will full schematics be released?

leeyu00, 2016-04-16:
good day, the Pi 3 schematic in the link below seems corrupted:
https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/schematics

pelwell, 2016-04-17:
Can you elaborate? It looks OK to me in Adobe Reader, OSX Preview and Chrome.

JamesH65, 2016-04-17:
Also fine in Chrome on Ubuntu 64bit.

leeyu00, 2016-04-17:
hi there, i tried downloading the file,"RPI-3B-V1_2-SCHEMATIC-REDUCED.pdf" and open in a pdf reader (Foxit 6.1.1) and it says it is corrupted. Upon inspection, the downloaded file and the file viewed online from here "https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/schematics/RPI-3B-V1_2-SCHEMATIC-REDUCED.pdf" has different file size. pardon for the trouble, must have done the checking beforehand. i have now downloaded the file by clicking the "raw" button on the link. thanks.

jeonghoonkang, 2016-04-18:
ㅍㅇ 투ㅡㅊ

-- 
Jeonghoon Kang
Managerial Researcher
at Korea Electronics Technology Institute(www.keti.re.kr)
TinyOS Korea Forum Administrator
www.tinyos.or.kr
work = budge@keti.re.kr
private = gadin.kang@gmail.com
https://github.com/jeonghoonkang
www.pinkfloyd.re.kr/rnd/
*Kmote Connector
http://www.tinyos.re.kr/bbs/upload/kmote_connector.pdf

lurch, 2016-04-18:
> i have now downloaded the file by clicking the "raw" button on the link.

If you want a "more direct" link (without github 'getting in the way'), you're probably better off accessing the documentation via https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md

nilnullzip, 2016-08-28:
The posted schematic is incomplete. Only a single page that does not have all the components. Is a more complete schematic available?

lurch, 2016-08-30:
> Is a more complete schematic available?

That's already answered by #347

awneil, 2016-08-30:
And the question was already asked on 12 April: https://github.com/raspberrypi/documentation/issues/321#issuecomment-208803605

commit 9abc8cf0 (2016-04-04): Add Pi 3 schematics and mechanical diagrams, close #321