raspberrypi/documentation #324: usb gadget on rpi A

prudy, 2016-03-11:
The rpi A could be used in device mode as the rpi Zero model. However as its otgid pin is permanently grounded it must force the HW/SW fixed device mode, not dual/otg.
Currently dwc2 driver gadget does not allow it as the gadget initialization overrides necessary settings.
Moreover, if dwc2 is compiled not in gadget-only but in dual-mode, the device mode must be forced with 'dtoverlay=dwc2,dr_mode=peripheral' in config.txt. Currently setting dr_mode parameter is ignored.
I'll issue the pull request fixing these, so gadgets with dwc2 work on raspberry A.

francesco-vannini, 2016-05-07:
Prudy is there any workaround to make the A+ work as a peripheral till this is fixed?

JamesH65, 2017-03-02:
@prudy Did you ever supply a PR for this? Cannot find one. Which part of the documentation are you suggesting needs updating?

prudy, 2017-03-06:
Hi, actually I did not, I solved the problem and sent a patch.
For the documentation - I think the mystery of otg pin routing should be described/explained. All settings go the DeviceTree thus this could be good place. If I'm not wrong some description was already there, the functionality was just broken.

JamesH65, 2017-03-07:
@prudy Do you have a PR number? Will make it easier to dig up.

prudy, 2017-03-07:
Only the patch mailing list story:
https://lists.gt.net/linux/kernel/2404917

JamesH65, 2017-03-15:
@ghollingworth @pelwell Is there any documentation covering this, if not should we have some. It's out of my knowledge base...

bennuttall, 2017-03-15:
http://blog.gbaman.info/?p=791

JamesH65, 2017-04-20:
So, what is actually required here? Do we need some documentation that parallels Ben's linked article?

bennuttall, 2017-04-21:
More info here: http://www.lucadentella.it/en/2017/04/10/raspberry-pi-zero-utilizzo-via-usb/

JamesH65, 2017-05-12:
I'd really like some confirmation on what needs/remains to be done here.  Is there actually something that needs doing, if so what, or can this issue be closed?

prudy, 2017-05-12:
James, I raised this ticket to start the process of pulling changes to the code. Sine this is already done in upstream there is no real need of having this opened.

JamesH65, 2017-05-12:
Ok, this is the documentation repo, so that has confused me. I'll close.

closed 2017-05-12 by JamesH65