raspberrypi/documentation #325: RaspberryPi 3 BCM2837 datasheet

pcotret, 2016-03-12:
Hi, does someone know if they will release the BCM2837 documentation soon ? I'm looking for the equivalent of the "BCM2835 ARM Peripherals"
https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf

I need to access the address of ARM components. I had a code on Pi2 but as Pi3 has a new CPU, I'm afraid these addresses may have changed.

pelwell, 2016-03-12:
That document still applies to the BCM2837, with the exception of ARM control block which is almost the same as that in the BCM2836.

lurch, 2016-03-12:
> I need to access the address of ARM components. I had a code on Pi2 but as Pi3 has a new CPU, I'm afraid these addresses may have changed.

#158 ?

noloader, 2016-07-31:
@pelwell. I believe the BCM2837 is ARMv8, while BCM2836 and below are ARMv7. BCM2837/ARMv8 has CRC32 instruction, while BCM2836/ARMv7 lack it.

(I'm here because I am looking for the other ARMv8 instructions, like PMULL, PMULL2, AES, SHA1, and SHA2. I'm trying to find the datasheet that details the presence/absence of the Crypto extensions).

pelwell, 2016-07-31:
As detailed in the [MagPi]( https://www.raspberrypi.org/magpi/raspberry-pi-3-specs-benchmarks/) and many other places, BCM2837 is a quad A53 clocked at 1.2GHz.

noloader, 2016-07-31:
@pelwell, The online magazines lack the details we need to determine some things. ARMv8 and the A53 core has optional extensions, like [CRC32 and Crypto](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0500g/CIHBIBBA.html). From [ARM Cortex-A53 MPCore Processor Technical Reference Manual, Section 4.3.20](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0500g/CIHBIBBA.html):

> The optional Cryptography engine is not included in the base product of the processor. ARM requires licensees to have contractual rights to obtain the Cortex-A53 Cryptography engine.

Most A53 cores I have come across have both of them. One early ARMv8 Mustang server board lacks both of them. RPI3 seems to fall into its own category - it has CRC32, but it lacks Crypto.

pelwell, 2016-07-31:
Any optional extension will also have a register to determine its presence. In the case of the crypto extensions it is ID_ISAR5.

noloader, 2016-07-31:
@pelwell, 

> Any optional extension will also have a register to determine its presence. In the case of the crypto extensions it is ID_ISAR5.

Sure, but it can't be read from user space because its EL1.

But what does that have to do with a data sheet on the SoC's specifications?

JamesH65, 2017-03-02:
See PR#527

JamesH65, 2017-03-03:
Fixed by #527

closed 2017-03-03 by JamesH65

kvu787, 2018-02-04:
I've found a BCM2837 datasheet linked from Stanford's CS 140e class. It does correctly map the peripherals to address 0x3Fnnnnnnn, unlike 0x20nnnnnn for the BCM2835. 
Other than that, I don't know how trustworthy it is.

See https://web.stanford.edu/class/cs140e/docs/BCM2837-ARM-Peripherals.pdf

lurch, 2018-02-04:
> Other than that, I don't know how trustworthy it is.

That's _very_ curious - it still has the copyright date of "06 February 2012" at the bottom of every page, and the string "2837" only appears once whilst the string "2835" appears 14 times. My guess would be that it's an unofficial modification of the original https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf

SergioBenitez, 2018-03-14:
@kvu787 @lurch I created that document. It is indeed simply a modified version of the original BCM2835 document with the intention of preventing confusion and bugs as students work through assignments. As explained [on the course website](https://web.stanford.edu/class/cs140e/assignments/0-blinky/#talking-to-hardware):

> If you open the manual we’ve linked you to, you’ll see references to the BCM2835 chip everywhere. This is because we’ve simply taken the documentation for the BCM2835 chip, fixed relevant errata [and addresses], and [changed] the title to say BCM2837.

lurch, 2018-03-14:
@SergioBenitez Ah cool, thanks for the info! Are you sure you're "allowed" to make (and publish) those changes though, given that the document clearly states "© 2012 Broadcom Corporation. All rights reserved" ?
\*shrug\*  I personally have no idea either way on the legalities, I just don't want to see anyone getting into trouble.

JamesH65, 2018-04-09:
This is being discussed in house - we are a bit short of the technical authors required to do a good job of it at the moment, and engeineeringwise are quite busy.

@ebenupton @ghollingworth