raspberrypi/documentation #326: If followed the instructions to update AVR Firmware and it bricked my Sense HAT

sisomm, 2016-03-14:
Hi I followed the instructions from point "EEPROM Data" Writing here 
https://www.raspberrypi.org/documentation/hardware/sense-hat/

And now it shows the rainbow colors permanently after bootup and no program can read or write to it. It seems completely bricked. What can I do???

[image]

XECDesign, 2016-03-14:
Could you upload your sense.eep somewhere?

Does it start working again if you add dtoverlay=rpi-sense to config.txt and reboot?

sisomm, 2016-03-14:
Hi, here's my sense.eep: https://dl.dropboxusercontent.com/u/2830831/sense.eep

And yes, adding that line to config.txt did the trick. thank you!

XECDesign, 2016-03-14:
Looks okay at a glance. Did eepflash.sh give any errors?

XECDesign, 2016-03-14:
Also, added a warning in https://github.com/raspberrypi/documentation/commit/c6e59ff8d386d7790959c7b0ded8a71fc8546fc5.

sisomm, 2016-03-14:
1. AFAIK eepflash.sh did not give errors until it stopped responding 
2. I get an errror running the command, while the shield works OK.. 

[image]

XECDesign, 2016-03-15:
Interesting, thanks. Could you post the output of `cat /proc/cpuinfo`?

sisomm, 2016-03-15:
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2709
Revision	: a02082
Serial		: 00000000f847f36d

sisomm, 2016-03-15:
We could skype & you could get keyboard control in an SSH session this afternoon if you want to peek around.

lurch, 2016-03-15:
> Revision : a02082

That indicates that you're using a Pi3B, and I [recently read](https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=137848&start=25#p917967) that **i2c-0** (which is what you need to read / write the EEPROM on a HAT board) can't be safely accessed from within Linux on a Pi3. Perhaps an additional check for this could/should be added to the eepflash tool?

sisomm, 2016-03-15:
Sounds like a Good idea. Btw is my sense hat now lost beyond repair?

sisomm, 2016-03-15:
Alternatively, can I salvage it by writing to it from at PI2B? I have a few laying about...

lurch, 2016-03-15:
> Alternatively, can I salvage it by writing to it from at PI2B?

From what I understand, I believe that should work.

@XECDesign Perhaps https://www.raspberrypi.org/documentation/hardware/sense-hat/ should be updated to clarify why / when you'd want to manually update the EEPROM on your Sense HAT?

sisomm, 2016-03-15:
I like to keep the firmware on my devices updated, as it normally makes them more stable. If the info given on the official raspberry site is likely to brick lots of devices... Whoa! But let's not jump to any conclusions?

sisomm, 2016-03-15:
UPDATE to @XECDesign and @lurch: I moved the SenseHat to a PI2B and did follow all the steps.. It worked like a charm. The hat is now back on the PI3B and works fine with the python examples the Eeprom read command doesn't work, though. You should be more specific in the warning and say that it doesn´t work with the PI3B...

XECDesign, 2016-03-16:
Clarified the warning in ecee854be6a219ff24039337aa8e4b1c77115ac1.

lurch, 2016-03-16:
> Clarified the warning in ecee854.

Shouldn't it also warn about the Pi 3B?

And IMHO "is are not be needed" doesn't make sense, the sentence would probably read better as "Please note that this operation is potentially dangerous and is not needed for the everyday user." ?

XECDesign, 2016-03-16:
Sorry, made the edit while busy with other things. Just an embarrassing editing errors. Thanks @lurch.

I was aware that the pi2 b rev1 used the i2c to talk to the SMPS, but that has since been done using another way. Just checked with the firmware/kernel folk and it turns out i2c0 is used for other things on the pi 3 b. Initially this was not a concern because it would either work or not touch the EEPROM at all, but on the pi 3, it seems that you can have i2c0 for a short while after booting, which is enough to be able to do minor damage.

Phil points out that is a bit-banged i2c interface that can be used instead which should get around all of this. I think I will test that and use it in the documentation instead. Thanks @sisomm for catching this problem.

LornaLynch, 2016-07-13:
Closing this as it appears to be resolved?

closed 2016-07-13 by LornaLynch