raspberrypi/documentation #331: config.txt: For HDMI_BOOST  add the acceptable values with description.

lwrbakro, 2016-03-29:
https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt.md

Please add the acceptable values for the HDMI_BOOST.
Mentioning each value's effect on the HDMI TMDS signal.

popcornmix, 2016-03-29:
The current documentation is not up to date.
config_hdmi_boost can go up to 11.
The default for a Pi1 model A or B is 2.
The default for Pi1 model B+ and all later boards (e.g. Pi0, Pi2, Pi3) is 5.

I think the default recommendation should be to try 7 if there are HDMI signal issues.
That can be increased further up to 11, but that should normally only be useful with very long HDMI cables.

JamesH65, 2017-03-02:
#526 Incorporates the changes suggested by @popcornmix

closed 2017-03-03 by JamesH65