raspberrypi/documentation #334: raspberry pi cant boot

aliisfandyar, 2016-03-31:
hello!
i powered raspberry pi with 7 volts,now active and power lights are always high and it cant boot
please help

lauraclay, 2016-03-31:
The Raspberry Pi should be powered by a 5V supply, as mentioned in the [FAQ](https://www.raspberrypi.org/help/faqs/#powerReqs) and the [documentation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md). Using a PSU with a higher voltage is not recommended, as it is very likely to damage the Pi beyond repair.

closed 2016-03-31 by lauraclay

lurch, 2016-03-31:
@lauraclay Looks like that documentation link you provided needs to be updated to cover the Pi 2B and Pi 3B as well?

lurch, 2016-03-31:
@aliisfandyar If you want to power your Raspberry Pi from a 7V source, you'll need to use a 5V step-down converter e.g. https://www.adafruit.com/products/1385

aliisfandyar, 2016-04-01:
will my rapberry pi will not work again :(

lauraclay, 2016-04-01:
@aliisfandyar Unlikely, if you ran 7V through it!