raspberrypi/documentation #337: 2837

bennuttall, 2016-04-05:
In [hardware/raspberrypi](https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi) we have [bcm2835](https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bcm2835) and [bcm2836](https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bcm2836). I guess we should add a 2837 folder with similar content.

ED6E0F17, 2016-06-05:
Looks like a five minute job. Aarch64 Crypto Instructions are an optional feature, so I am assuming it doesn`t have those, but you need to Document that choice.

machinaut, 2016-12-18:
Have the relevant documents been released by broadcom?  Happy to send a pull request if that's the case.

I was under the impression that they weren't released yet.

pelwell, 2016-12-18:
It would be much quicker just to try building with the crypto instructions
and see if it works.

Ruffio, 2017-01-07:
@bennuttall can this issue be closed?

JamesH65, 2017-03-02:
New page has been  added for some time. Closing.

closed 2017-03-02 by JamesH65