raspberrypi/documentation #340: Remote desktop to PI through web browser - does not work with JESSIE

windy54, 2016-04-08:
Instructions in the Teachers caossroon guide do not work with JESSIE.
Step 1: setting up server on PI
 paragraphs 6,7,8 need replacing with issue 237 

Step 2: needs a new system file similar to Issue 237.
 I have created the following file /etc/systemd/system/vncproxy.service

[Unit]
Description=Remote desktop service via web broswer
After=rsyslog.service,network.target

[Service]
Type=simple
RemainAfterExit=yes
ExecStart=/usr/bin/sudo /usr/bin/screen -S noVNC -dms noVNC /usr/local/share/noVNC/utils/launch.sh --vnc localhost:5901 --listen 80
[Install]
WantedBy=multi-user.target

then enabled it with
sudo systemctl enable "vncproxy.service"

This appears to work, i.e. limited testing at home,I got there with a lot of trial and error, I am not confident enough with linux to say this is the correct solution!

UnderFact, 2016-10-03:
the new update to jessie has built in vnc server and viewer software

windy54, 2016-10-03:
Yes I have tried it an can access using vnc software on my laptop. On wheezey I could access the PI from a browser without having to install any software on my laptop and that is what I have been trying to recreate and am struggling with.

lurch, 2016-10-03:
Given that the page below has 'archive' in the URL, I wonder if this feature is no longer supported in current versions of RealVNC? @minkustree ?
https://archive.realvnc.com/products/vnc/documentation/4.0/win/java.html

minkustree, 2016-10-03:
@lurch The Java viewer is supported in VNC 5.3.x, although we're working on a replacement technology since fewer and fewer browsers support the necessary Java plugin any more.

However, I don't think this is how @windy54 is trying to connect - it looks like noVNC is being used to connect to a virtual-mode VNC server, and then provide a web-based proxy to a VNC session. This mechanism *should* work, but is not something we routinely test against. It may require tweaks to the commands needed to launch vncserver, as the command-line options for the RealVNC version have evolved from the tightvnc and other versions based on the original RealVNC open-source implementation.

I can't see anything in the [teacher's guide](https://www.raspberrypi.org/learning/teachers-guide/remote/) about accessing a VNC Server from the web - is this documented elsewhere?

windy54, 2016-10-03:
It was documented in the previous version of the teachers guide .

Ruffio, 2017-01-07:
@windy54 can this issue be closed?

JamesH65, 2017-03-02:
Since this refers to an older version of the Teachers guide, I am going to close it. Please reopen if anyone thinks there is still an issue.

closed 2017-03-02 by JamesH65