raspberrypi/documentation #343: Power supply page is ambiguous and not updated to latest product lines

P33M, 2016-04-09:
https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/power

Two points are being addressed in this page that should probably be separated
- How to power your Raspberry Pi
- How to power components attached to your Raspberry Pi

Each require updating to align with Pi2/Pi3 in any case.

lurch, 2016-04-09:
And on a related note, the stuff about TP1 and TP2 should probably be removed from https://github.com/raspberrypi/documentation/blob/master/installation/sd-cards.md and https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt.md

LornaLynch, 2016-07-13:
We've edited these documents and made some changes. Please reopen this if there are still problems.

closed 2016-07-13 by LornaLynch