raspberrypi/documentation #345: raspivid page missing mp4/playback/encoding information

dtbaker, 2016-04-10:
The raspivid page was my first point of contact when I was trying to use this new command. I successfully followed the instructions and saved a video output, but that video output wouldn't play (or would play incorrectly) in mplayer/VLC. 

There is no information about encoding or playback options for this raw saved raspivid output. 

I suggest adding some basic information about the output file and some options on how to play that file back on a computer. Here is some basic information with a basic mp4 example:

https://github.com/raspberrypi/documentation/compare/master...dtbaker:patch-1?diff=split&name=patch-1

(have not issued a pull request on that patch as per the contrib guidelines which say open an issue first)

Cheers,
Dave

JamesH65, 2016-04-11:
That patch looks fine - please submit it, then Ben or similar can review
and merge.

JamesH65, 2017-03-06:
@dtbaker Any chance you can submit this as a PR?

dtbaker, 2017-03-06:
@JamesH65 no worries, updated here: https://github.com/raspberrypi/documentation/pull/346

closed 2017-03-06 by dtbaker