raspberrypi/documentation #347: Raspberry Pi 3 FULL Schematics

awneil, 2016-04-12:
Only "_reduced_" schematics - little more than connector pinouts - were released to close #321 

Will **_full_** schematics be released?

lurch, 2016-04-12:
I doubt it - see also:  https://github.com/raspberrypi/documentation/search?q=schematics&type=Issues

bennuttall, 2016-05-26:
It's unlikely anything else will be released. What do you need?

awneil, 2016-05-26:
Why is it unlikely?

I want the full schematics so that I don't have to keep coming back and asking for the bit(s) I don't currently have each time I realise that the bit I need at a particular time is not available.

Whatever happened to "Open Source" ?

bennuttall, 2016-05-26:
Raspberry Pi has never been *open hardware*. All our software is open source, and we release as much of the hardware schematics as we can.

closed 2016-05-26 by bennuttall

awneil, 2016-05-26:
Fair point.

So what is actually preventing your from releasing at least as much of the schematics as was available for the original Pi ??

Why are you making work for yourselves by cutting out "non release" stuff?

JamesH65, 2016-05-26:
Because it IS work (quite a lot) to release schematics, work that for the
huge majority of people is completely irrelevant? And even old Pi models
are still relevant, for example the Zero is based on the PI1.

Anyway, this sort of discussion is better on the forum, where there are
already threads on exactly this subject.

The answer is still the same though.

konakcc, 2016-06-14:
Can we get at least a block diagram with all the devices and IO map?  I heard rumor that the SD card slot has limitations.

lotsacaffeine, 2016-08-30:
Would you consider releasing the full schematics under an NDA?

JamesH65, 2016-08-31:
Just out of interest, what information do you need that the schematic
provides - perhaps that information can be provided in another way?

LizUpton, 2016-08-31:
I'm afraid we do not release schematics under NDA unless there are very, very special circumstances (i.e. you're building Raspberry Pis under licence).

E3V3A, 2016-09-14:
For one good reason, we may be able to use the schematics to be able to use the RPi3B as an USB OTG gadget device, as I mentioned [here](https://github.com/raspberrypi/documentation/issues/282#issuecomment-247076992), in issue #282

sigboe, 2016-09-22:
@LizUpton @bennuttall Can I please have a description of the test pads? I specifically want to relocate the SD card using test pads like I can on the Pi0.

gtsiros, 2016-09-23:
Yeah so i bought a B yesterday and just checked the schematics.

Very limited information in there and one could be forgiven for thinking that what was released was the bare minimum to satisfy #321

ppumkin, 2017-01-09:
> Because it IS work (quite a lot) to release schematics, work that for the
> huge majority of people is completely irrelevant? And even old Pi models
> are still relevant, for example the Zero is based on the PI1.

The only hard part of releasing the schematic is taking arranging the schematics into the reduced version. When you design the PCB, like say using Eagle PCB the schematics are there, auto generated. Its not like you doing it by hand? If you are .. you are doing it wrong. LOL

I want to see how the WiFi/BT is connected.. to which bus etc. Even a simple block diagram would be usefull so we can use that on other forums.. to explain to new comers how things are connected.

But without factual information from the foundation.. allot of people just come up with crapware answers. 

I do not see how a logical block diagram would infringe on any contractual obligations you have made with all those big boys. We just would like to know how things are connected officially - Without second guessing the whole time.

Thanks

pelwell, 2017-01-09:
> I want to see how the WiFi/BT is connected.. to which bus etc. Even a simple block diagram would be usefull so we can use that on other forums.. to explain to new comers how things are connected.

That information is published in the Device Tree: WiFi is connected via SDIO on GPIOs 34-39, while Bluetooth is connected to UART0 on GPIOs 32&33.

awneil, 2017-01-09:
> The only hard part of releasing the schematic is taking arranging the schematics into the reduced version. 

My thoughts exactly!

stevedee, 2017-02-08:
Is there any information on the Pi3 USB part of the circuit? (I can't see it in the reduced schematic).

My Pi3 USB has just failed as there is no 5V on the connectors, so I'm only interested in tracing this part of the supply circuit.

pklapperich, 2017-04-13:
Considering using the CM3 or CM3L in a project, but we'll be wanting to add wifi, 2 ethernet, and bluetooth. Since the RPI3 has all of these things, it would have been really convenient to look at how things are connected on the RPI3 schematic. But the reduced schematic is basically useless. The Ethernet port isn't even shown, and all that's shown for the HDMI port is the connector itself with some signal names. ... I already know the signal names for an HDMI port; that's standardized.

claymation, 2017-04-17:
I would like to use the Raspberry Pi for a project teaching how to translate schematics to devicetree, but the reduced schematics don't include even the simplest nets -- like which GPIO the status LED is connected to.

Anyone who's going to do nefarious things with the board can reverse the schematic, anyway, so you're not preventing the bad guys from doing bad things, you're just preventing the good guys from trying to do good things.

JamesH65, 2017-04-17:
There are currently no intentions to release the full schematics of the Pi
range, except the CM. So I'd suggest using the CM for your device tree
lesson plan. Since it actually exposes more interfaces than the rest of the
range, it's probably a better choice.

pklapperich, 2017-04-17:
Claymation, check out Pine64. They have full schematics for all of their products, even their upcoming laptop. We're currently evaluating their SoDIMM form factor product for our project that we had originally intended to use CM3 for.

JamesH65, 2017-04-18:
Any particular reason why the CM is not suitable?

awneil, 2017-04-18:
There's also the various Beagle boards ...

https://beagleboard.org/

JamesH65, 2017-04-18:
Since this thread is fast becoming an advertising stream for other peoples boards, is there any way to stop commenting?

JamesH65, 2017-04-18:
Yes, there is....